PrincetonUniversity/openpiton #2: openpiton/piton/tools/Linux/x86_64/m4_gmp does not work out of box

jrrk, 2018-11-30:
The mktools command recompiles certain utilities such as goldfinger, however there is no mention of m4_gmp which is a mystery binary. This silently fails unless libc6-i386 is installed and no meaningful error is given. Also libgmp10:i386 is a dependency which is not mentioned. Once this is sorted, simulation cannot work unless libbit-vector-perl is installed (these package names refer to Ubuntu 16.04), a necessary prerequisite to eventually use Vivado.

Perhaps a sanitised version of this workaround could be added to the getting started instructions.

Jbalkind, 2018-11-30:
Thank you for bringing this up!

There's an intended fix for this discussed (esoterically) in the FPGA manual on page 2. Does piton/tools/src/proto/syscfg.sh still work to install the necessary packages?

The problem here is that we're still using the specially-produced version of M4 that includes built-in multi-precision support (and is compiled into a single binary) that came with the OpenSPARC T1 distribution, because some T1 stuff relies on mpeval(). I built a newer version for Mac but haven't got round to it for Linux as I was worried about just introducing similar new portability problems. I'll move this up my priority list.

getziadz, 2019-01-11:
Same problem on Fedora release 29.
Resolved by installing gmp.i686 and glibc.i686
# sudo dnf install gmp.i686 glibc.i686

Jbalkind, 2019-06-04:
The main issue should be resolved and the README.md updated to explain the dependencies with the merge of #3

closed 2019-06-04 by Jbalkind