PrincetonUniversity/openpiton #14: Latch inference in dynamic_node

Jbalkind, 2019-03-29:
We've been getting some timing issues when pushing high frequencies on the new UltraScale+ FPGAs due to latch inference in a few parts of the design. @msfschaffner has fixed most of these but the one that we're uncertain about and want a good long-term solution to is `dynamic_node`. @msfschaffner has implemented several alternatives to the latch-inferring block from dynamic_node, `one_of_five`, and collected some data from synthesising that one block for ASIC.

@wentzlaf we discussed using the "zeroes" version of the patch. @msfschaffner has since implemented another option which is "array" - it's much better with respect to area and slack for ASIC synthesis, but the "error case" for this is left up to the tool. If we trust that DC will know that the design only uses the first 5 values, then the error case shouldn't be hit anyway.

We should decide which is best for the long term and go with that. @wentzlaf which is your preference?

Here are the synthesis results:

```
+---------+---------+---------+---------+
| variant | [squmm] | [ps]    | ts [ps] |    
+---------+---------+---------+---------+
| latch   | 2824.4  | 338.46  | 4252.8  | 
+---------+---------+---------+---------+
| zeroes  | 2615.5  | 341.40  | 11029.2 | 
+---------+---------+---------+---------+
| bypass  | 2626.0  | 351.62  | 16204.2 |  
+---------+---------+---------+---------+
| array   | 2280.2  | 353.19  | 5091.3  |  
+---------+---------+---------+---------+
```

The implementations are as follows (default `latch` version of the file is [here](https://github.com/PrincetonUniversity/openpiton/blob/openpiton/piton/design/chip/tile/dynamic_node/components/rtl/one_of_five.v)):

```// latch
module one_of_five_latch(in0,in1,in2,in3,in4,sel,out);
    parameter WIDTH = 8;
    parameter BHC = 10;
    input [2:0] sel;
    input [WIDTH-1:0] in0,in1,in2,in3,in4;
    output reg [WIDTH-1:0] out;
    always@(*)
    begin
        case(sel)
            3'd0:out=in0;
            3'd1:out=in1;
            3'd2:out=in2;
            3'd3:out=in3;
            3'd4:out=in4;
            default:; // indicates null
        endcase
    end
endmodule

// bypass
module one_of_five_bypass(in0,in1,in2,in3,in4,sel,out);
    parameter WIDTH = 8;
    parameter BHC = 10;
    input [2:0] sel;
    input [WIDTH-1:0] in0,in1,in2,in3,in4;
    output reg [WIDTH-1:0] out;
    always@(*)
    begin
        out=in0;
        case(sel)
            3'd0:out=in0;
            3'd1:out=in1;
            3'd2:out=in2;
            3'd3:out=in3;
            3'd4:out=in4;
            default:; // indicates null
        endcase
    end
endmodule

// zeroes
module one_of_five_zeroes(in0,in1,in2,in3,in4,sel,out);
    parameter WIDTH = 8;
    parameter BHC = 10;
    input [2:0] sel;
    input [WIDTH-1:0] in0,in1,in2,in3,in4;
    output reg [WIDTH-1:0] out;
    always@(*)
    begin
        out={WIDTH{1'b0}};
        case(sel)
            3'd0:out=in0;
            3'd1:out=in1;
            3'd2:out=in2;
            3'd3:out=in3;
            3'd4:out=in4;
            default:; // indicates null
        endcase
    end
endmodule

// array
module one_of_five_array(in0,in1,in2,in3,in4,sel,out);
    parameter WIDTH = 8;
    parameter BHC = 10;
    input [2:0] sel;
    input [WIDTH-1:0] in0,in1,in2,in3,in4;
    output [WIDTH-1:0] out;
    wire [5*WIDTH-1:0] tmp;
    assign tmp ={in4,in3,in2,in1,in0};
    assign out = tmp[sel*WIDTH +: WIDTH];
endmodule```

wentzlaf, 2019-03-29:
I would prefer the "zero" implementation.

David

Jbalkind, 2019-03-29:
Let's go with `zero` then. @msfschaffner, would you mind submitting a PR referencing this issue for us to run our CI on?

Jbalkind, 2019-04-10:
With #15 I see the following:

Synthesis for `dynamic_node` passes (no new warnings/errors, timing is met) so we should be good. Area is pretty different. Output of the area report is below.

New design:

```Number of ports:                          693
Number of nets:                          5948
Number of cells:                         5515
Number of combinational cells:           4290
Number of sequential cells:              1194
Number of macros/black boxes:               0
Number of buf/inv:                        752
Number of references:                     448

Combinational area:               4897.736976
Buf/Inv area:                      662.103002
Noncombinational area:            5134.077067
Macro/Black Box area:                0.000000
Net Interconnect area:      undefined  (No wire load specified)

Total cell area:                 10031.814043
Total area:                 undefined

Core Area:                     16101
Aspect Ratio:                 1.0002
Utilization Ratio:            0.6231
```

Old design:

```Number of ports:                          693
Number of nets:                          5902
Number of cells:                         5469
Number of combinational cells:           3920
Number of sequential cells:              1519
Number of macros/black boxes:               0
Number of buf/inv:                        754
Number of references:                     294

Combinational area:               3624.075009
Buf/Inv area:                      351.819004
Noncombinational area:            5501.808086
Macro/Black Box area:                0.000000
Net Interconnect area:      undefined  (No wire load specified)

Total cell area:                  9125.883094
Total area:                 undefined

Core Area:                     15354
Aspect Ratio:                 1.0046
Utilization Ratio:            0.5943
```

msfschaffner, 2019-04-10:
This is interesting... which parameterizations of `one_of_*.v` are used in this particular instance of `dynamic_node`? One further option we may explore is using don't cares for the unused states (this would be similar to the array implementation above, but without arrays), as that implementation seemed to produce the smallest design (just looking at the muxes).

(Also referencing #15)

msfschaffner, 2019-04-17:
Ok I get exactly the opposite result for FDX22.

old design:
```
Number of ports:                          922
Number of nets:                          7544
Number of cells:                         7019
Number of combinational cells:           4316
Number of sequential cells:              2654
Number of macros/black boxes:               0
Number of buf/inv:                        495
Number of references:                     220

Combinational area:               1433.369603
Buf/Inv area:                       73.482240
Noncombinational area:            4645.555204
Macro/Black Box area:                0.000000
Net Interconnect area:      undefined  (No wire load specified)

Total cell area:                  6078.924808
Total area:                 undefined
```

new design:
```
Number of ports:                          922
Number of nets:                          7172
Number of cells:                         6647
Number of combinational cells:           4269
Number of sequential cells:              2329
Number of macros/black boxes:               0
Number of buf/inv:                        496
Number of references:                     213

Combinational area:               1428.377603
Buf/Inv area:                       74.813440
Noncombinational area:            4320.675853
Macro/Black Box area:                0.000000
Net Interconnect area:      undefined  (No wire load specified)

Total cell area:                  5749.053456
Total area:                 undefined
```

Maybe this is tool-version or technology dependent...

closed 2019-05-21 by msfschaffner