PrincetonUniversity/openpiton #32: dtc tool needed but doesn't seem to be in the requirements?

mithro, 2019-06-13:
Following the instructions in the `README.md`;
```sh
cd $PITON_ROOT/
source piton/ariane_setup.sh
piton/ariane_build_tools.sh
cd $PITON_ROOT/build
sims -sys=manycore -x_tiles=1 -y_tiles=1 -msm_build -ariane
```

You get the following output;
```
install -p -m 644 `find isa -maxdepth 1 -type f` /tmp/riscv-tests/build/share/riscv-tests/isa                                  
install -p -m 644 `find benchmarks -maxdepth 1 -type f` /tmp/riscv-tests/build/share/riscv-tests/benchmarks                   
                                                                                                              
----------------------------------------------------------------------                                                        
build complete                                                                                                            
----------------------------------------------------------------------                                                   
                                                                                                                      
tansell@tansell:~/github/PrincetonUniversity/openpiton$ cd $PITON_ROOT/build
tansell@tansell:~/github/PrincetonUniversity/openpiton/build$ sims -sys=manycore -x_tiles=1 -y_tiles=1 -msm_build -ariane
/usr/local/google/home/tansell/github/PrincetonUniversity/openpiton/piton/design/chip/tile/ariane/Flist.ariane
compiling DTS and bootroms for Ariane...
rm -f bootrom.img bootrom.sv ariane.dtb
dtc -I dts ariane.dts -O dtb -o ariane.dtb
make: dtc: Command not found
Makefile:20: recipe for target 'ariane.dtb' failed
make: *** [ariane.dtb] Error 127
sims: Caught a SIG Error compiling DTS for ariane. at /usr/local/google/home/tansell/github/PrincetonUniversity/openpiton/piton/tools/src/sims/sims,2.0 line 1246.
```

mithro, 2019-06-13:
Probably need a `sudo apt-get install device-tree-compiler`

mithro, 2019-06-13:
After installing `device-tree-compiler` I now get to a missing `vlog` command;
```
sims: Building rtl model
sims: vlog -mfcu  -suppress 2583 -f flist 
Can't exec "vlog": No such file or directory at /usr/local/google/home/tansell/github/PrincetonUniversity/openpiton/piton/tools/src/sims/sims,2.0 line 1545.
sims: Caught a SIGDIE. failed building model at /usr/local/google/home/tansell/github/PrincetonUniversity/openpiton/piton/tools/src/sims/sims,2.0 line 1546.
```

wzc314, 2020-02-20:
> After installing `device-tree-compiler` I now get to a missing `vlog` command;
>
> ```
> sims: Building rtl model
> sims: vlog -mfcu  -suppress 2583 -f flist
> Can't exec "vlog": No such file or directory [...]

I've met the same problem: Can't exec "vlog"
Have you solved it?

Jbalkind, 2020-02-20:
`vlog` is the unhelpful name for ModelSim(/Questa). You'll need to run a different command from `-msm_build` and `-msm_run` if you want to use a different simulator.

I thought this was described in the readme but I see it's not. Sorry about that. We do note it in the tutorial slides that you can find on openpiton.org, if you want a bit more of a "getting started" guide.

`-vcs_build` and `-vcs_run` for Synopsys VCS
`-ncv_build` and `-ncv_run` for Cadence NCSim (Incisive)
`-msm_build` and `-msm_run` for Mentor ModelSim (Questa)
`-vlt_build` and `-vlt_run` for Verilator
`-icv_build` and `-icv_run` for Icarus Verilog
`-riv_build` and `-riv_run` for Aldec Riviera Pro

Note that you'll need to use a simulator that supports SystemVerilog if you want to use the Ariane core.