PrincetonUniversity/openpiton #34: VERILATOR_ROOT does not work with install verilator

mithro, 2019-06-13:
Ran `./configure --prefix=/opt/verilator` and did a `make install`. Then set `VERILATOR_ROOT` to `/opt/verilator` and got the following error;
```
i_fmt_slice.gen_num_lanes[0].active_lane.lane_instance.i_fma.output_pipline.i_output_pipe.stage_ready
sims: make -j -C /usr/local/google/home/tansell/github/PrincetonUniversity/openpiton/build/manycore/rel-0.1/obj_dir -f Vcmp_top.mk Vcmp_top
make: Entering directory '/usr/local/google/home/tansell/github/PrincetonUniversity/openpiton/build/manycore/rel-0.1/obj_dir'
Vcmp_top.mk:65: /opt/verilator/include/verilated.mk: No such file or directory
make: *** No rule to make target '/opt/verilator/include/verilated.mk'.  Stop.
make: Leaving directory '/usr/local/google/home/tansell/github/PrincetonUniversity/openpiton/build/manycore/rel-0.1/obj_dir'
```
The make file is actually in `/opt/verilator/share/verilator/include/verilated.mk`.

If I instead set `VERILATOR_ROOT` to the source directory, everything works.

hughiexi, 2022-03-02:
I met the same problem as you except I install Verilator globally. What do you mean "set VERILATOR_ROOT to the source directory"?

Jbalkind, 2022-03-02:
VERILATOR_ROOT is a path. If you set its value to the location of the source code (i.e. the directory you get when you untar the tarball), then it should seemingly clear the issue.

hughiexi, 2022-03-03:
Thanks for your answer!