PrincetonUniversity/openpiton #50: Building Ariane+Openpiton Bitstream for Genesys2

genesys2, 2019-12-11:
I have issues generating the Ariane+Openpiton bitstream for a Genesys2 board. 
I followed the README, that is:

1. Setting PITON_ROOT
2. Ran source $PITON_ROOT/piton/piton_settings.bash
3. cd $PITON_ROOT/
4. source piton/ariane_setup.sh
5. piton/ariane_build_tools.sh

Okay, so far, so good.

Now i want to run `protosyn -b genesys2 -d system --core=ariane --uart-dmw ddr` which crashes with the following output: 
```
[INFO]  protosyn,2.5:702: ----- System Configuration -----
[INFO]  protosyn,2.5:720: x_tiles   = 1
[INFO]  protosyn,2.5:721: y_tiles   = 1
[INFO]  protosyn,2.5:722: num_tiles = 1
[INFO]  protosyn,2.5:729: core      = ariane
[INFO]  protosyn,2.5:758: setenv RTL_ARIANE0
[INFO]  protosyn,2.5:775: network   = 2dmesh_config
[INFO]  protosyn,2.5:779: l15 size  = 8192
[INFO]  protosyn,2.5:780: l15 assoc = 4
[INFO]  protosyn,2.5:781: l1d size  = 8192
[INFO]  protosyn,2.5:782: l1d assoc = 4
[INFO]  protosyn,2.5:783: l1i size  = 16384
[INFO]  protosyn,2.5:784: l1i assoc = 4
[INFO]  protosyn,2.5:785: l2  size  = 65536
[INFO]  protosyn,2.5:786: l2  assoc = 4
[INFO]  protosyn,2.5:800: ---- Additional RTL Defines ----
[INFO]  protosyn,2.5:803: NO_RTL_CSM
[INFO]  protosyn,2.5:803: PITON_FPGA_MC_DDR3
[INFO]  protosyn,2.5:803: PITONSYS_MEM_ZEROER
[INFO]  protosyn,2.5:803: PITON_FPGA_SD_BOOT
[INFO]  protosyn,2.5:803: PITONSYS_UART_BOOT
[INFO]  protosyn,2.5:803: PITON_NO_CHIP_BRIDGE
[INFO]  protosyn,2.5:803: PITON_UART16550
[INFO]  protosyn,2.5:803: PITON_FPGA_ETHERNETLITE
[INFO]  protosyn,2.5:805: --------------------------------
[INFO]  protosyn,2.5:874: Generating UART init sequence
[INFO]  protosyn,2.5:631: Using core clock frequency: 66.667 MHz
[INFO]  protosyn,2.5:285: Building a project for design 'system' on board 'genesys2'
[INFO]  protosyn,2.5:330: Running FPGA implementation down to bitstream generation
[INFO]  protosyn,2.5:927: Checking Project Build results
[ERROR] fpga_lib.py:344: Vivado Project was not created properly!
[ERROR] fpga_lib.py:345: Check: /home/jan/Desktop/openpiton/build/genesys2/system/protosyn_logs/make_project.log
```

The logfile contains the following hint: 
```
INFO: compiling DTS and bootroms for Ariane (MAX_HARTS=1, UART_FREQ=66667000)...
INFO: done
INFO: generating PLIC for Ariane (2 targets, 2 sources)...
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00007f5cabe03740 (most recent call first):
    while executing
"exec ./gen_plic_addrmap.py -t $NUM_TARGETS -s $NUM_SOURCES > plic_regmap.sv"
    invoked from within
"if  {[info exists ::env(PITON_ARIANE)]} {
  puts "INFO: compiling DTS and bootroms for Ariane (MAX_HARTS=$::env(PTON_NUM_TILES), UART_FREQ=$env(CONFIG..."
    (file "/home/jan/Desktop/openpiton/piton/tools/src/proto/common/setup.tcl" line 114)

    while executing
"source $DV_ROOT/tools/src/proto/common/setup.tcl"
    (file "/home/jan/Desktop/openpiton/piton/tools/src/proto/vivado/setup.tcl" line 30)

    while executing
"source $DV_ROOT/tools/src/proto/vivado/setup.tcl"
    (file "/home/jan/Desktop/openpiton/piton/tools/src/proto/vivado/gen_project.tcl" line 33)
INFO: [Common 17-206] Exiting Vivado at Wed Dec 11 13:48:21 2019...
```

I have checked my python installation and removed the environment variables PYTHONHOME and PYTHONPATH and also tried running the command in screen envrironment which was suggested by google. Btw, the python script runs just fine when called separately. Now, I do realize that this might very well be an issue with the setup of my Environment but I have no idea what to try next. So if anyone has an idea, I would appreciate your help. 

Thank you.

Jbalkind, 2019-12-11:
Hi there,

I think this was addressed in a discussion on our [google group](https://groups.google.com/forum/#!forum/openpiton) at the end of October. The newer Vivado is setting its own Python environment inside the setup.tcl script as part of our build flow.

If you add the following:

```
unset ::env(PYTHONPATH)    
unset ::env(PYTHONHOME)  
```

between the `puts` and the `cd` here: https://github.com/PrincetonUniversity/openpiton/blob/7c2f5aa6038a8b8384de4dc6aadf94bf3da4602b/piton/tools/src/proto/common/setup.tcl#L131-L133

and give it a try again then you may see success.

Thanks!
Jon

genesys2, 2019-12-12:
Hi Jon,

Thanks alot, this seems to work. Protosyn is now running for quite a while wihtout crashing so I am confident that it will succeed.

Best,
Jan

closed 2020-04-16 by Jbalkind

shirlynan, 2021-01-06:
I met this problem agian. 
However, add 
unset ::env(PYTHONPATH)    
unset ::env(PYTHONHOME)  
didn't solve the problem...

Jbalkind, 2021-01-06:
Credit to Ivan Cheung for a solution here:

```
I write down here as a ref for others...
add following 4 lines at line 113 of setup.tcl to store PYTHONPATH and  PYTHONHOME  environment variables. 
 set tmp_PYTHONPATH $env(PYTHONPATH)
 set tmp_PYTHONHOME $env(PYTHONHOME)
  unset ::env(PYTHONPATH)    
  unset ::env(PYTHONHOME)
and add the following 2 lines at line 120 of setup.tcl to restore the environment... 
 set ::env(PYTHONPATH) $tmp_PYTHONPATH
 set ::env(PYTHONHOME) $tmp_PYTHONHOME
```

shirlynan, 2021-01-07:
> set tmp_PYTHONPATH $env(PYTHONPATH)
>  set tmp_PYTHONHOME $env(PYTHONHOME)
>   unset ::env(PYTHONPATH)    
>   unset ::env(PYTHONHOME)

Thanks to your reply. I've tried with this solution however without success.
`# Pre-process PyHP files
source $DV_ROOT/tools/src/proto/common/pyhp_preprocess.tcl
set ALL_RTL_IMPL_FILES [pyhp_preprocess ${ALL_RTL_IMPL_FILES}]
set ALL_INCLUDE_FILES [pyhp_preprocess ${ALL_INCLUDE_FILES}]

set tmp_PYTHONPATH $env(PYTHONPATH)
set tmp_PYTHONHOME $env(PYTHONHOME)
  unset ::env(PYTHONPATH)    
  unset ::env(PYTHONHOME)

if  {[info exists ::env(PITON_ARIANE)]} {
  puts "INFO: compiling DTS and bootroms for Ariane (MAX_HARTS=$::env(PITON_NUM_TILES), UART_FREQ=$env(CONFIG_SYS_FREQ))..." 
  set TMP [pwd]
  cd $::env(ARIANE_ROOT)/openpiton/bootrom/baremetal
  # Note: dd dumps info to stderr that we do not want to interpret
  # otherwise this command fails...

  set ::env(PYTHONPATH) $tmp_PYTHONPATH
  set ::env(PYTHONHOME) $tmp_PYTHONHOME

  exec make clean 2> /dev/null`

This is where I inserted the code. And I can run both python and python3 on my machine.

`Python 3.5.2 (default, Oct  7 2020, 17:19:02) 
[GCC 5.4.0 20160609] on linux

Python 2.7.12 (default, Oct  5 2020, 13:56:01) 
[GCC 5.4.0 20160609] on linux2
`

Jbalkind, 2021-01-07:
Could you share the error messages you see?

shirlynan, 2021-01-07:
> Could you share the error messages you see?
> 
> Also are you sure that those are the right lines? I don't see anything between those two lines that would be using the pythonpath...

The line numbers are indeed the ones you mentioned. But I also realized that they are not in the right position. I thought line 107 and line 116 should be the right ones.
The error message is shown below. 
```
INFO: compiling DTS and bootroms for Ariane (MAX_HARTS=1, UART_FREQ=100000000)...
no such variable
    (read trace on "env(PYTHONPATH)")
    invoked from within
"set tmp_PYTHONPATH $env(PYTHONPATH)"
    invoked from within
"if  {[info exists ::env(PITON_ARIANE)]} {
  puts "INFO: compiling DTS and bootroms for Ariane (MAX_HARTS=$::env(PITON_NUM_TILES), UART_FREQ=$env(CONFI..."
    (file "/home/openpiton/piton/tools/src/proto/common/setup.tcl" line 125)

    while executing
"source $DV_ROOT/tools/src/proto/common/setup.tcl"
    (file "/home/openpiton/piton/tools/src/proto/vivado/setup.tcl" line 30)

    while executing
"source $DV_ROOT/tools/src/proto/vivado/setup.tcl"
    (file "/home/openpiton/piton/tools/src/proto/vivado/gen_project.tcl" line 33)
INFO: [Common 17-206] Exiting Vivado at Wed Jan  6 21:58:24 2021...
```
When I removed 
```
set ::env(PYTHONPATH) $tmp_PYTHONPATH
set ::env(PYTHONHOME) $tmp_PYTHONHOME
```
from line 116, this error seems disappeared. The project is currently running, and I hope that no new errors will occur. Thanks for your help.