PrincetonUniversity/openpiton #59: benchmarking ine ariane core in openpiton

AnouarNechi, 2020-03-27:
Hi I want to run dhrystone and Coremark on  Ariane core  using XCVU9P (existing FPGA in AWS F1 instance)
Is it possible to run open piton with only 1 Ariane core and run the benchmarks?
Can I configure the L1-I and L1-D sizes?
Are the RS, BTB, and BHT configurable for each Ariane core?
Thank you

Jbalkind, 2020-03-30:
You can certainly build with just a single core and change those parameters as necessary. We and our users routinely test this way.

For L1 caches, keep in mind that the L1.5 should be inclusive of the L1-D. You can specify cache sizes to protosyn when generating the FPGA bitfile if you check the arguments with `protosyn -h`.

For the Ariane predictor parameters, you'll need to change those in the Ariane RTL directly for each instantiation.

AnouarNechi, 2020-04-02:
I ran these commands for the moment/
`$ export PITON_ROOT=`pwd``
`$ source piton/ariane_setup.sh`
`$ piton/ariane_build_tools.sh`
But I am not seeing the Ariane RTL files to change the RAS BTB and BHT parameters.
where can I find them.

Jbalkind, 2020-04-02:
This is tied to your issue in #61 where the submodule isn't being checked out. Try the openpiton-dev branch instead. Also don't run ariane_setup and ariane_build_tools without configuring them for your environment first. If you already have a riscv cross compiler for example, you can point ariane_setup to it (or change the path where it will be built). Further, if you have one, then you can also remove the line in ariane_build_tools that compiles you a new cross compiler.

closed 2020-04-04 by AnouarNechi