PrincetonUniversity/openpiton #63: error whilev building simulation model

AnouarNechi, 2020-04-04:
I want to run benchmark on simulation so I did the fillowing;
```
$ source $PITON_ROOT/piton/piton_settings.bash
$ source piton/ariane_setup.sh
$ piton/ariane_build_tools.sh
$ cd $PITON_ROOT/build
$ sims -sys=manycore -x_tiles=1 -y_tiles=1 -vlt_build -ariane
```
and I got the following error
```
sims -sys=sys_ariane -x_tiles=1 -y_tiles=1 -vlt_build -ariane 
sims: ====================================================
sims:   Simulation Script for OpenPiton
sims:   Modified by Princeton University on June 9th, 2015
sims: ====================================================
sims: ====================================================
sims:   Simulation Script for OpenSPARC T1
sims:   Copyright (c) 2001-2006 Sun Microsystems, Inc.
sims:   All rights reserved.
sims: ====================================================
sims: start_time Sat Apr  4 18:54:30 CEST 2020
sims: running on nechi
sims: uname is Linux nechi 5.3.0-45-generic #37~18.04.1-Ubuntu SMP Fri Mar 27 15:58:10 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
sims: version 2.0
sims: dv_root /home/nechi/Desktop/openpiton/piton
sims: model_dir /home/nechi/Desktop/openpiton/build
sims: tre_search /home/nechi/Desktop/openpiton/piton/tools/env/tools.iver
sims: using config file /home/nechi/Desktop/openpiton/piton/tools/src/sims/sims.config ()
Can't exec "bw_cpp": No such file or directory at /home/nechi/Desktop/openpiton/piton/tools/src/sims/sims,2.0 line 2034.
sims: Caught a SIGDIE. Could not open /home/nechi/Desktop/openpiton/piton/tools/src/sims/sims.config at /home/nechi/Desktop/openpiton/piton/tools/src/sims/sims,2.0 line 2034.
```
what could be the problem?

Jbalkind, 2020-04-04:
From the readme:

```If you see an error with bw_cpp then you may need to install gcc/g++ (to get cpp), or csh (csh on ubuntu, tcsh on centos)```

I would recommend looking for/asking some of these kinds of questions on our [google group](https://groups.google.com/forum/#!forum/openpiton). Someone else had this same issue there the other day too.

AnouarNechi, 2020-04-04:
I solved that issue it was sourcing issue.
Now when I run 
`$ sims -sys=manycore -msm_run -x_tiles=1 -y_tiles=1 dhrystone.riscv -ariane -precompiled`
 I get this error:
```
sims -sys=manycore -msm_run -x_tiles=1 -y_tiles=1 dhrystone.riscv -ariane -precompiled 
sims: ====================================================
sims:   Simulation Script for OpenPiton
sims:   Modified by Princeton University on June 9th, 2015
sims: ====================================================
sims: ====================================================
sims:   Simulation Script for OpenSPARC T1
sims:   Copyright (c) 2001-2006 Sun Microsystems, Inc.
sims:   All rights reserved.
sims: ====================================================
sims: start_time Sat Apr  4 19:56:05 CEST 2020
sims: running on nechi
sims: uname is Linux nechi 5.3.0-45-generic #37~18.04.1-Ubuntu SMP Fri Mar 27 15:58:10 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
sims: version 2.0
sims: dv_root /home/nechi/Desktop/openpiton/piton
sims: model_dir /home/nechi/Desktop/openpiton/build
sims: tre_search /home/nechi/Desktop/openpiton/piton/tools/env/tools.iver
sims: using config file /home/nechi/Desktop/openpiton/piton/tools/src/sims/sims.config ()
sims: looking for precompiled RISCV binary dhrystone.riscv
sims: using random seed 0
network_config not specified, assuming 2dmesh configuration
Setting UART_DIV_LATCH to 0xb
sims: locating diag dhrystone.riscv
sims: Looking for diag under $DV_ROOT/verif/diag
sims: Looking for diag under /home/nechi/Desktop/openpiton/piton/design/chip/tile/ariane//tmp/riscv-tests/build/isa
sims: Looking for diag under /home/nechi/Desktop/openpiton/piton/design/chip/tile/ariane//tmp/riscv-tests/build/benchmarks
sims: Found diag under /home/nechi/Desktop/openpiton/piton/design/chip/tile/ariane//tmp/riscv-tests/build/benchmarks/dhrystone.riscv
sims: rv64_img
truncate diag.o -s +48 
sims: Caught a SIGDIE. work/ not found in /home/nechi/Desktop/openpiton/build/manycore/rel-0.1 at /home/nechi/Desktop/openpiton/piton/tools/src/sims/sims,2.0 line 1614.
```

`

Jbalkind, 2020-04-04:
We don't use Modelsim a lot in our team but I think you might have missed the -msm_build step first?

Jon

Jbalkind, 2020-04-04:
Then you also want -vlt_run, not -msm_run