PrincetonUniversity/openpiton #64: JTAG Support VCU118

shirlynan, 2020-04-04:
Hello everyone,
I want to know which kind of JTAG debugger is supported on VCU118 board.
Xilinx internal JTAG chain via the bscane2 primitive or external JTAG debugger?
Is any changes needed on fpga/ariane.cfg under ariane repo?

Thank you all.

Jbalkind, 2020-04-04:
For the Genesys2 board, we've primarily used the external JTAG debugger. I think for VCU118 and some other boards we would have to use bscane2. We haven't used it yet (though the wiring is there) so we don't have the modified ariane.cfg figured out just yet. We'd love to work with you to help figure it out though because it would be useful on many other boards, including on Amazon F1.

You can see where it would be instantiated here: https://github.com/PrincetonUniversity/openpiton/blob/openpiton-dev/piton/design/rtl/system.v#L610-L639

AnouarNechi, 2020-04-05:
I uncommented the bscane2 section in system.v as instructed in the previous comment and initiated the generation of the bitstream but I got the following error:
```
Phase 1.2 IO Placement/ Clock Placement/ Build Placer Device
ERROR: [Place 30-1100] Failed to do pre-placement. Reason: Cannot place instance dbg_hub/inst/BSCANID.u_xsdbm_id/SWITCH_N_EXT_BSCAN.bscan_inst/SERIES7_BSCAN.bscan_inst of type BSCANE2 in site CONFIG_SITE_X0Y1 with message Instance dbg_hub/inst/BSCANID.u_xsdbm_id/SWITCH_N_EXT_BSCAN.bscan_inst/SERIES7_BSCAN.bscan_inst can not be placed in BSCAN1 of site CONFIG_SITE_X0Y1 because the bel is occupied by BSCANE2_inst. This could be caused by bel constraint conflict
Phase 1.2 IO Placement/ Clock Placement/ Build Placer Device | Checksum: 22fea2ae

Time (s): cpu = 00:00:02 ; elapsed = 00:00:02 . Memory (MB): peak = 6095.594 ; gain = 0.000 ; free physical = 2472 ; free virtual = 10505
Phase 1 Placer Initialization | Checksum: 22fea2ae

Time (s): cpu = 00:00:02 ; elapsed = 00:00:02 . Memory (MB): peak = 6095.594 ; gain = 0.000 ; free physical = 2472 ; free virtual = 10505
ERROR: [Place 30-99] Placer failed with error: 'Placer initialization failed'
Please review all ERROR, CRITICAL WARNING, and WARNING messages during placement to understand the cause for failure.
Ending Placer Task | Checksum: cfcf3e29

Time (s): cpu = 00:00:02 ; elapsed = 00:00:02 . Memory (MB): peak = 6095.594 ; gain = 0.000 ; free physical = 2473 ; free virtual = 10506
INFO: [Common 17-83] Releasing license: Implementation
86 Infos, 52 Warnings, 13 Critical Warnings and 3 Errors encountered.
place_design failed
ERROR: [Common 17-69] Command failed: Placer could not place all instances
INFO: [Common 17-206] Exiting Vivado at Mon Apr  6 00:04:31 2020...
```
How can I fix this error please?

Jbalkind, 2020-04-06:
As I say, we haven't tried this out yet. If it's saying the instance is in use, then I might guess that you may need to change the value for JTAG_CHAIN? I'm not sure.

shirlynan, 2020-04-07:
> For the Genesys2 board, we've primarily used the external JTAG debugger. I think for VCU118 and some other boards we would have to use bscane2. We haven't used it yet (though the wiring is there) so [...]

I've tried but with no luck. I also tried to modify the openocd configuration file for VCU118, but It keeps reporting errors like
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: riscv.cpu: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unsupported DTM version: 15
Error: Target not examined yet

Error: Unsupported DTM version: 15

Is there any tips?
Thanks

Jbalkind, 2020-04-09:
All I have are some pointers I was given previously but haven't had the chance to follow up on. The PR adding support to OpenOCD has since been included, so make sure it's new enough first: https://github.com/riscv/riscv-openocd/pull/345

The same bscane2 is used by lowRISC (which also uses Ariane) who have a short guide on their own usage: https://www.lowrisc.org/docs/jtag/

I think if you look at the two of those it should give you a rough idea of what you'll need to change in the ariane config file. To be clear: you recompiled with bscane2 included, right? - did you change the chain? Did you hit the same issue as Anouar above?

shirlynan, 2020-04-09:
> All I have are some pointers I was given previously but haven't had the chance to follow up on. The PR adding support to OpenOCD has since been included, so make sure it's new enough first: [riscv/riscv-openocd#345](https://github.com/riscv/riscv-openocd/pull/345)
> 
> The same bscane2 is used by lowRISC (which also uses Ariane) who have a short guide on their own usage: https://www.lowrisc.org/docs/jtag/
> 
> I think if you look at the two of those it should give you a rough idea of what you'll need to change in the ariane config file. To be clear: you recompiled with bscane2 included, right? - did you change the chain? Did you hit the same issue as Anouar above?

Thanks, I'll have take a look.
Yes, I recompiled the project to make bscane2 included. and I also change the chain number. Because the debug core of ddr4 has taken the first register.
one more question, so it means that the external FTDI JTAG debugger is no more supported on VCU118 board, right? Or should I attempt both approaches?

Jbalkind, 2020-04-09:
I think on Genesys2 we can use the FT2232H JTAG chip's second channel because there's a second chain wired up to the FPGA for that channel. For VCU118 I can't see that chain wired up in the XDC at least so I think we need to connect to the first chain and that that means we have to use the bscane2. I'm not an expert here, though. This is roughly what I remember from Michael who set up the JTAG for Genesys2 and who wrote that bscane2 instantiation which was commented out due to lack of testing.

jctullos, 2020-08-15:
@shirlynan Did you ever get yours working?

So I can get openocd to recognize the board, but gives DTM version unsupported. From doing this on other builds, it usually meant I needed to have the board reset and wait for a connection from openocd. Is there a specific configuration we need to build to wait for a debug connection? Right now it's built for the sd boot which doesn't work.

maskedarray, 2023-02-20:
Hi, @jctullos and @shirlynan were you able to make it work?

jimaandro, 2024-07-17:
Hello !
I am trying to use OpenOCD with VCU too. Did anyone managed to make this work?

Jbalkind, 2024-07-21:
You might try checking out the recently opened PR for several Alveo boards which apparently has BSCAN based JTAG: https://github.com/PrincetonUniversity/openpiton/pull/150