PrincetonUniversity/openpiton #67: running a program on AWS f1 instance

AnouarNechi, 2020-04-07:
Hello I have read the instructions to work with openpiton on AWS f1 instance and I have some questions:
when is it possible to run a precompiled benchmark is necessary to be executed before uploading the OS or it does not matter?
to run a precompiled benchmark for example dhrystone can I use this instruction:
`pitonstream -b f1 -d system dhrystone.riscv --core=ariane --precompiled` ?
From where do I get the linux image to upload to openpiton design on F1 instance. What do you mean by  **$FILE_LOCATION**  in `./dma_os $FILE_LOCATION`.

grigoriy-chirkov, 2020-04-09:
Hi Anouar!

1. If you just want to run benchmark  - you dont have to load OS, you can run it with pitonstream
2. Yes, this should work. But you also need to specify the UART port in the command with -p flag, e.g. "-p pts/6"
3. You can download it from  our website https://parallel.princeton.edu/openpiton/ .
Hope this helps:)

Best, 
Grigory

AnouarNechi, 2020-04-10:
Thank you Grigory
In order to see the UART should I open another console to see the execuation or the pitonstream command will open it automatically.