PrincetonUniversity/openpiton #70: How to increase the ariane core frequency

AnouarNechi, 2020-04-13:
I want to increase the ariane core frequency ... How can I do that?
what is the max frequency reachable ?

Jbalkind, 2020-04-13:
That's entirely dependent on the environment. If you have an UltraScale+ VU9P part (like the vcu118 or Amazon F1) you should be able to reach 125MHz. On Kintex 7 (Digilent Genesys 2) we get about 67MHz and on Virtex 7 (VC707) we get about 60MHz.

If you run a `protosyn` build once, then you can open the project in the Vivado GUI and reconfigure the `clk_mmcm` and `uart16550` IPs to change the clock frequencies for those. You'll also want to update `piton/tools/src/proto/block.list` to set the new frequency for `system` on your chosen board. I think that's it but there may be another IP or something. Perhaps `atg_uart_init`?

AnouarNechi, 2020-04-14:
I kept changing the frequency in piton/tools/src/proto/block.list file and I reached 165 MHz until now and did not reach timing violation is this normal?

Jbalkind, 2020-04-14:
Did you also regenerate the IPs for the new frequency? `clk_mmcm` is the one generating the clock at the new frequency. `block.list` only helps generate the bootrom and so on.

AnouarNechi, 2020-04-14:
I assume the  clk_out_1 is responsible for the core right?

Jbalkind, 2020-04-14:
Should be, yeah