PrincetonUniversity/openpiton #77: NVDLA Accelerator Inference Simulation

Hassan313, 2020-06-04:
How can we use OpenPiton to do inference simulation with NVDLA accelerator? Is it possible to do it without FPGA?

Jbalkind, 2020-06-08:
At present we don't have a great simulation-only environment for this. For FPGA, we can make use of AXI-based infrastructure to hook up NVDLA on a crossbar to memory, but to do this in simulation we don't presently have an AXI-based fake memory. It don't think it would be too hard to extend our fake memory to do something like this, though.

Another approach some students took was to translate from the AXI memory port of NVDLA to P-Mesh messages. However, this effort wasn't implemented to completion.

redpanda3, 2025-05-16:
Hi, Professor Jonathan, how's this project going now? For an AXI accelerator with memory, a 'good' option is to connect over the chipset, and for the accelerators near processor, a 'good' option is connect over the noc router within the chip. Correct?  Here are the  solutions that I thought to connect accelerators

[image]

Jbalkind, 2025-05-16:
Right, exactly, there are several options, though not exactly according to your rendering. We've done the chipset one (NVDLA speaks directly to DRAM, non-coherently, but the DRAM is still shared with the rest of the system) and the NoC one (NVDLA speaks to the L2 coherently via the NoC with an AXI->P-Mesh bridge). We have not done connection of something like NVDLA via the L1.5 TRI described in the BYOC paper, but Duet (HPCA 2023) does use TRI for connecting an FPGA fabric.