PrincetonUniversity/openpiton #81: Building Ariane+Openpiton Bitstream for Genesys2  is  Failed

pclshancheng, 2020-06-24:
I followed the README, that is:
Setting PITON_ROOT
Ran source $PITON_ROOT/piton/piton_settings.bash
cd $PITON_ROOT/
source piton/ariane_setup.sh
piton/ariane_build_tools.sh

when I run protosyn -b genesys2 -d system --core=ariane --uart-dmw ddr which crashes with the following output:
**protosyn -> .local_tool_wrapper: configsrch returned error code 126 Exiting ...**
What caused it and how to debug, if anyone has an idea, I would appreciate your help.

Jbalkind, 2020-06-24:
It's been a while since I last saw a configsrch error so I'm trying to think through the usual causes. Perhaps we can go back and forth a little just to make sure we identify the root cause to better document for future.

Firstly, does running `mktools` help? I think that should rebuild configsrch

Jbalkind, 2020-06-24:
Other things to check:
- Do you have a `csh` or `tcsh`?
- Do you have python (both 2 & 3 if you're working from the `openpiton` branch, just 3 if you're working from `openpiton-dev`)?
- Do you have the packages listed in ariane_setup.sh (or equivalents for your distro)?

Alexmojito, 2020-06-29:
> Other things to check:
> 
> * Do you have a `csh` or `tcsh`?
> * Do you have python (both 2 & 3 if you're working from the `openpiton` branch, just 3 if you're working from `openpiton-dev`)?
> * Do you have the packages listed in ariane_setup.sh (or equivalents for your distro)?

thanks，after the packages listed in ariane_setup.sh was installed, and the problem was sovled.

closed 2020-06-29 by Alexmojito