PrincetonUniversity/openpiton #82: Implementation is failed

Alexmojito, 2020-06-29:
when i run protosyn -b genesys2 -d system --core=ariane --uart-dmw ddr which crashes with the following output:
[INFO]  protosyn,2.5:702: ----- System Configuration -----
[INFO]  protosyn,2.5:720: x_tiles   = 1
[INFO]  protosyn,2.5:721: y_tiles   = 1
[INFO]  protosyn,2.5:722: num_tiles = 1
[INFO]  protosyn,2.5:729: core      = ariane
[INFO]  protosyn,2.5:732: defining RTL_TILE0
[INFO]  protosyn,2.5:762: setenv RTL_ARIANE0
[INFO]  protosyn,2.5:780: network   = 2dmesh_config
[INFO]  protosyn,2.5:784: l15 size  = 8192
[INFO]  protosyn,2.5:785: l15 assoc = 4
[INFO]  protosyn,2.5:786: l1d size  = 8192
[INFO]  protosyn,2.5:787: l1d assoc = 4
[INFO]  protosyn,2.5:788: l1i size  = 16384
[INFO]  protosyn,2.5:789: l1i assoc = 4
[INFO]  protosyn,2.5:790: l2  size  = 65536
[INFO]  protosyn,2.5:791: l2  assoc = 4
[INFO]  protosyn,2.5:805: ---- Additional RTL Defines ----
[INFO]  protosyn,2.5:808: NO_RTL_CSM
[INFO]  protosyn,2.5:808: PITON_FPGA_MC_DDR3
[INFO]  protosyn,2.5:808: PITONSYS_MEM_ZEROER
[INFO]  protosyn,2.5:808: PITON_FPGA_SD_BOOT
[INFO]  protosyn,2.5:808: PITONSYS_UART_BOOT
[INFO]  protosyn,2.5:808: PITON_NO_CHIP_BRIDGE
[INFO]  protosyn,2.5:808: PITON_UART16550
[INFO]  protosyn,2.5:808: PITON_FPGA_ETHERNETLITE
[INFO]  protosyn,2.5:810: --------------------------------
[INFO]  protosyn,2.5:879: Generating UART init sequence
[INFO]  protosyn,2.5:631: Using core clock frequency: 66.667 MHz
[INFO]  protosyn,2.5:285: Building a project for design 'system' on board 'genesys2'
[INFO]  protosyn,2.5:330: Running FPGA implementation down to bitstream generation
[INFO]  protosyn,2.5:932: Checking Project Build results
[INFO]  fpga_lib.py:348: Project was build successfully!
[INFO]  protosyn,2.5:939: Checking Project Implementation results
Traceback (most recent call last):
  File "/home/shancheng/PITON_ROOT/piton/tools/src/proto/protosyn,2.5", line 949, in <module>
    main()
  File "/home/shancheng/PITON_ROOT/piton/tools/src/proto/protosyn,2.5", line 940, in main
    if not implFlowSuccess(rc_dir.log, rc_dir.run):
  File "/home/shancheng/PITON_ROOT/piton/tools/src/proto/fpga_lib.py", line 365, in implFlowSuccess
    if not strInFile(fpath, ["synth_design completed successfully"]):
  File "/home/shancheng/PITON_ROOT/piton/tools/src/proto/fpga_lib.py", line 330, in strInFile
    f = open(fpath, 'r')
FileNotFoundError: [Errno 2] No such file or directory: '/home/shancheng/PITON_ROOT/build/genesys2/system/genesys2_system/genesys2_system.runs/synth_1/runme.log'

so i dont konw  why runme.log was deleted, how to find the reason， if anyone has an idea, please help me .

Jbalkind, 2020-06-29:
Huh, this is strange. Can you check the other logs (vivado.log, protosyn_logs/) under build/genesys2/system/? Did the synthesis run actually start? Which Vivado version are you using?

Alexmojito, 2020-06-30:
> Huh, this is strange. Can you check the other logs (vivado.log, protosyn_logs/) under build/genesys2/system/? Did the synthesis run actually start? Which Vivado version are you using?

thanks, the synthesis has already run, and the  Vivado version is 2018.3.the vivado.log and protosyn_logs have reported the warning :
【 WARNING: [IP_Flow 19-3664] IP 'clk_mmcm' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/xilinx/genesys2/ip_cores/clk_mmcm/clk_mmcm.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'clk_mmcm' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/xilinx/genesys2/ip_cores/clk_mmcm/clk_mmcm_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'clk_mmcm' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/xilinx/genesys2/ip_cores/clk_mmcm/clk_mmcm_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'clk_mmcm' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/xilinx/genesys2/ip_cores/clk_mmcm/clk_mmcm_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'clk_mmcm' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/xilinx/genesys2/ip_cores/clk_mmcm/clk_mmcm_sim_netlist.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'afifo_w64_d128_std' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/xilinx/genesys2/ip_cores/afifo_w64_d128_std/afifo_w64_d128_std.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'afifo_w64_d128_std' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/xilinx/genesys2/ip_cores/afifo_w64_d128_std/afifo_w64_d128_std_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'afifo_w64_d128_std' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/xilinx/genesys2/ip_cores/afifo_w64_d128_std/afifo_w64_d128_std_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'afifo_w64_d128_std' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/xilinx/genesys2/ip_cores/afifo_w64_d128_std/afifo_w64_d128_std_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'afifo_w64_d128_std' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/xilinx/genesys2/ip_cores/afifo_w64_d128_std/afifo_w64_d128_std_sim_netlist.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'mac_eth_axi_lite' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/xilinx/genesys2/ip_cores/mac_eth_axi_lite/mac_eth_axi_lite.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'mac_eth_axi_lite' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/xilinx/genesys2/ip_cores/mac_eth_axi_lite/mac_eth_axi_lite_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'mac_eth_axi_lite' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/xilinx/genesys2/ip_cores/mac_eth_axi_lite/mac_eth_axi_lite_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'mac_eth_axi_lite' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/xilinx/genesys2/ip_cores/mac_eth_axi_lite/mac_eth_axi_lite_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'mac_eth_axi_lite' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/xilinx/genesys2/ip_cores/mac_eth_axi_lite/mac_eth_axi_lite_sim_netlist.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'atg_uart_init' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/io_ctrl/xilinx/genesys2/ip_cores/atg_uart_init/atg_uart_init.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'atg_uart_init' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/io_ctrl/xilinx/genesys2/ip_cores/atg_uart_init/atg_uart_init_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'atg_uart_init' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/io_ctrl/xilinx/genesys2/ip_cores/atg_uart_init/atg_uart_init_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'atg_uart_init' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/io_ctrl/xilinx/genesys2/ip_cores/atg_uart_init/atg_uart_init_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'atg_uart_init' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/io_ctrl/xilinx/genesys2/ip_cores/atg_uart_init/atg_uart_init_sim_netlist.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'uart_16550' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/io_ctrl/xilinx/genesys2/ip_cores/uart_16550/uart_16550.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'uart_16550' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/io_ctrl/xilinx/genesys2/ip_cores/uart_16550/uart_16550_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'uart_16550' generated file not found '/home/shancheng/PITON_ROOT/piton/design/chipset/io_ctrl/xilinx/genesys2/ip_cores/uart_16550/uart_16550_stub.vhdl'. Please regenerate to continue.
[... cut ...]  added.
CRITICAL WARNING: [Vivado 12-1462] The source file is already part of the fileset 'sources_1'. Requested source '/home/shancheng/PITON_ROOT/piton/design/common/fpga_bridge/fpga_rcv/xilinx/genesys2/ip_cores/afifo_w3/afifo_w3.xci' will not be added.】
so that may be the reason,how can i solve it,thanks
[implementation.log](https://github.com/PrincetonUniversity/openpiton/files/4849090/implementation.log)
[vivado.log](https://github.com/PrincetonUniversity/openpiton/files/4849091/vivado.log)

Jbalkind, 2020-06-30:
Did you by any chance change Vivado version while you've been trying to get protosyn to work? I'd suggest deleting `build/genesys2` and retrying. If that doesn't work, let me know what files you see under `piton/design/chipset/xilinx/genesys2/ip_cores/clk_mmcm` for example

Alexmojito, 2020-06-30:
thanks ,the vivado version didn't change ,and  I try to run  by  deleting build/genesys2,but it still  can not work ,so the file under piton/design/chipset/xilinx/genesys2/ip_cores/clk_mmcm is as follows：
mmcm_pll_drp_func_us_plus_pll.vh
mmcm_pll_drp_func_us_plus_mmcm.vh
mmcm_pll_drp_func_us_pll.vh
mmcm_pll_drp_func_us_mmcm.vh
mmcm_pll_drp_func_7s_pll.vh
mmcm_pll_drp_func_7s_mmcm.vh
clk_mmcm_ooc.xdc
clk_mmcm_clk_wiz.v
clk_mmcm_board.xdc
clk_mmcm.xml
clk_mmcm.xdc
clk_mmcm.xci
clk_mmcm.veo
clk_mmcm.v

Jbalkind, 2020-06-30:
Hmm ok. Could you try the following: `find piton/design/ | grep genesys2/ip_cores | grep "\.xml$" | xargs rm`

It looks kind of like Vivado created the synthesis jobs for the mmcm, brams, fifos, etc but then didn't actually run them to successful completion. I've been able to get a bit more success from it in the past when I removed any .xml and .dcp files from the genesys2/ip_cores/ subdirectories before.

Let me know if this helps!

Alexmojito, 2020-06-30:
thanks, I try to remove this xml and .dcp files many times, but it still donsen't work .

Jbalkind, 2020-06-30:
Ok could you give it a try on a clean clone of the repo? Now that you have the other dependencies set up maybe that would help?

It really looks like it doesn't run the synthesis jobs for the IPs. One thing you could also try is to open the genesys2_system.xpr in the Vivado GUI and run synthesis of the IPs as out-of-context jobs yourself.

Alexmojito, 2020-07-01:
thanks, the problem have been  found  when i open the genesys2_system.xpr  in the Vivado GUI. the  IP license has expired, so I  apply the new license,and the Implementation is passed.

Jbalkind, 2020-07-01:
Ah that's interesting. Good to note for future! I'll close this for now :)

closed 2020-07-01 by Jbalkind