PrincetonUniversity/openpiton #86: Running proxy kernel on Openpiton

Rahul-Kande, 2020-07-28:
Hello, 

May I know if there is a way to run proxy kernel on openpiton (https://github.com/riscv/riscv-pk) through simulation and through running openpiton on AWS F1 instances?
I am planning on running simple user applications on top of the proxy kernel on openpiton.

Rahul-Kande, 2020-07-28:
does this work? 
```
< compile pk >
mv <pk elf file> <piton/build path>
mv pk diag.exe
rv64_img
<vsim command>
```

Jbalkind, 2020-07-31:
We don't have support for the proxy kernel built in, though stubs of what you would need to implement for use in simulation are there. Michael worked on this a little before he finished his work on the Ariane integration, but it wasn't complete and I don't personally know what more is needed as I haven't used the proxy kernel myself.

You can see the RISCV_FESVR_SIM macro here: https://github.com/openhwgroup/cva6/blob/dcea6c97d43b4a03f703f515990b3eab9082e6c9/openpiton/riscv_peripherals.sv#L112-L139

If you just want to be able to do debug with gdb, you could definitely wire the top level JTAG in system.v to whatever virtual JTAG that is provided by F1. JTAG works correctly on the genesys2 so if the F1 virtual JTAG is similar then it should be straightforward. Using a BSCANE2-based JTAG like some of the other boards would use (https://github.com/PrincetonUniversity/openpiton/blob/0fd0985dc8140094fff6693a679abdcd3596a4c6/piton/design/rtl/system.v#L619-L639) would require some more tinkering with openocd.

Is there something in particular that the bare metal and Linux environments don't give that you want from the pk? We haven't had much interest since Linux is working reliably.