PrincetonUniversity/openpiton #94: Ariane Toolchain is misssing when compiling the benchmarks

AnouarNechi, 2021-01-13:
I am trying to have a fresh installation of openpitonand I id the following:
```
$ git clone <openpiton repo>
$ git checkout -b openpiton-dev
$ export PITON_ROOT=path to openpiton
$ source $PITON_ROOT/piton/piton_settings.bash
$ cd  $PITON_ROOT
$ source piton/ariane_setup.sh
$ piton/ariane_build_tools.sh
```
At this point I get an error during the benchmarks compilation :

```
make[1]: riscv64-unknown-elf-gcc: Command not found
make[1]: *** [/home/rk-vcu118/Desktop/openpiton/piton/design/chip/tile/ariane/tmp/riscv-tests/build/../benchmarks/Makefile:54: median.riscv] Error 127
make[1]: Leaving directory '/home/rk-vcu118/Desktop/openpiton/piton/design/chip/tile/ariane/tmp/riscv-tests/build/benchmarks'
make: *** [Makefile:25: benchmarks] Error 2
```
How could I fix this issue. Thank you

Jbalkind, 2021-01-13:
Your ariane_build_tools.sh likely failed but you missed the error message.

Two suggestions:
- Check the dependencies listed in the file
- Modify ariane_setup.sh to set the RISCV environment variable to point to a location that you want to install riscv gcc to.

AnouarNechi, 2021-01-13:
Totally right the RISCV env. variable was wrong and needs to be edited. Now I can see that the Toolchain is cloned and being installed.
Thank you Jona

closed 2021-01-13 by AnouarNechi