PrincetonUniversity/openpiton #96: Test compilation failed

RaphaelKlink, 2021-01-14:
Hey,

when i try to execute the pitonstream command for the dhrystone benchmark i get this error:

command:
`pitonstream -b vcu118 -d system -f ./piton/design/chip/tile/ariane/tmp/riscv-tests/build/benchmarks/dhrystone.riscv  --core=ariane -precompiled -p ttyUSB1 --no_wait_fpga_config`

console output:
```
dhrystone.c
[INFO]  pitonstream,1.0:372: UART DIV Latch value: 0x36
[INFO]  pitonstream,1.0:375: Configuring port /dev/ttyUSB1
[INFO]  pitonstream,1.0:164: UART will be configured for 115200 baud rate
[INFO]  pitonstream,1.0:402: Running dhrystone.c: 1 out of 1 test
[INFO]  pitonstream,1.0:287: Compiling dhrystone.c
sims -sys=manycore -novcs_build -midas_only               -midas_args='-DUART_DIV_LATCH=0x36 -DFPGA_HW -DCIOP -DNO_SLAN_INIT_SPC' dhrystone.c -ariane -uart_dmw -x_tiles=1 -y_tiles=1
[ERROR] pitonstream,1.0:409: Test compilation failed
[ERROR] pitonstream,1.0:410: Skipping dhrystone.c
[ERROR] pitonstream,1.0:411: See /home/rk-vcu118/Desktop/openpiton/build/uart_piton.log for more information
```

Any help to solve this problem

Jbalkind, 2021-01-14:
What does the log that it tells you to check say?

RaphaelKlink, 2021-01-14:
sims -sys=manycore -novcs_build -midas_only -midas_args="-DUART_DIV_LATCH=0x36 -DFPGA_HW -DCIOP -DNO_SLAN_INIT_SPC" dhrystone.c -ariane -uart_dmw -x_tiles=1 -y_tiles=1 
sims: ====================================================
sims:   Simulation Script for OpenPiton
sims:   Modified by Princeton University on June 9th, 2015
sims: ====================================================
sims: ====================================================
sims:   Simulation Script for OpenSPARC T1
sims:   Copyright (c) 2001-2006 Sun Microsystems, Inc.
sims:   All rights reserved.
sims: ====================================================
sims: start_time Do 14. Jan 18:18:37 CET 2021
sims: running on vcu118-vm
sims: uname is Linux vcu118-vm 5.8.0-36-generic #40~20.04.1-Ubuntu SMP Wed Jan 6 10:15:55 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
sims: version 2.0
sims: dv_root /home/rk-vcu118/Desktop/openpiton/piton
sims: model_dir /home/rk-vcu118/Desktop/openpiton/build
sims: tre_search /home/rk-vcu118/Desktop/openpiton/piton/tools/env/tools.iver
sims: using config file /home/rk-vcu118/Desktop/openpiton/piton/tools/src/sims/sims.config ()
sims: using random seed 0
network_config not specified, assuming 2dmesh configuration
sims: locating diag dhrystone.c
sims: Looking for diag under $DV_ROOT/verif/diag
sims: Caught a SIGDIE. no diag with name dhrystone.c found at /home/rk-vcu118/Desktop/openpiton/piton/tools/src/sims/sims,2.0 line 2439.

Jbalkind, 2021-01-14:
My guess would be that you haven't prebuilt the riscv tests in the ariane subrepo if it can't find dhrystone. Perhaps your running of ariane_build_tools.sh failed and you didn't notice. You can comment out some of the steps near the top so it doesn't rebuild gcc if you don't want to wait for that build to repeat

RaphaelKlink, 2021-01-15:
Hello 
I double checked the steps you mentioned and it seems that the benchmarks are well compiled and I already found the executables where they should be.
[image]

Jbalkind, 2021-01-15:
Ah then I think you might need to put dhrystone.riscv rather than .c

RaphaelKlink, 2021-01-18:
I am on the newest commit of the development branch.
I have tried the .riscv as well as the .c

When i put the .c file it get this output

`rk-vcu118@vcu118-vm:~/Desktop/openpiton$ pitonstream -b vcu118 -d system -f ./piton/design/chip/tile/ariane/tmp/riscv-tests/benchmarks/dhrystone/dhrystone.c  --core=ariane -precompiled -p ttyUSB1 --no_wait_fpga_config
[INFO]  pitonstream,1.0:372: UART DIV Latch value: 0x36
[INFO]  pitonstream,1.0:375: Configuring port /dev/ttyUSB1
[INFO]  pitonstream,1.0:164: UART will be configured for 115200 baud rate

==============================
All tests finished

Exiting...
`
and the uart_piton.log is completely empty

The command log for the compiled .riscv is written in my inital text

Jbalkind, 2021-01-18:
Sanity check: in both cases you put -precompiled rather than --precompiled - which are you using in your command? It should be the latter. I don't see sims printing that you gave the precompiled argument and its output indicates it's not looking in the location it should be (set by the precompiled flag) for precompiled riscv tests for ariane.

Note: pitonstream uses --precompiled (because python) while sims uses -precompiled (because perl).

RaphaelKlink, 2021-01-19:
I tried it with -precompiled and --precompiled and it failed both times with the same error

Jbalkind, 2021-01-19:
Wait your commands point to the file you're trying to run (`-f ./piton/design/chip/tile/ariane/tmp/riscv-tests/benchmarks/dhrystone/dhrystone.c`). What you should be doing for pitonstream is creating a file called test.txt (`-f test.txt`) which has the names of the test you want to run (not paths, so just `dhrystone.c`). I hadn't noticed this before. sims will search for the file for you, you don't specify the path.

RaphaelKlink, 2021-01-24:
yes that was the error thank you for your help. And apparently i am not on the dev branch for some reason when i checkout the dev branch out, i accidently created a copy of the master branch with the name openpiton-dev.

I have another problem now. I want to lower the frequenz for the vcu118 to 50 MHZ, but I get no uart output for the tests:

'rk-vcu118@vcu118-vm:~/Desktop/openpiton$ pitonstream -b vcu118 -d system -f ./piton/design/chip/tile/ariane/ci/riscv-benchmarks.list --core=ariane --precompiled -p ttyUSB1 --no_wait_fpga_config
dhrystone.riscv
median.riscv
multiply.riscv
pmp.riscv
qsort.riscv
rsort.riscv
towers.riscv
vvadd.riscv
[INFO]  pitonstream,1.0:372: UART DIV Latch value: 0x36
[INFO]  pitonstream,1.0:375: Configuring port /dev/ttyUSB1
[INFO]  pitonstream,1.0:164: UART will be configured for 115200 baud rate
[INFO]  pitonstream,1.0:402: Running dhrystone.riscv: 1 out of 8 test
[INFO]  pitonstream,1.0:287: Compiling dhrystone.riscv
sims -sys=manycore -novcs_build -midas_only               -midas_args='-DUART_DIV_LATCH=0x36 -DFPGA_HW -DCIOP -DNO_SLAN_INIT_SPC' dhrystone.riscv -ariane -uart_dmw -x_tiles=1 -y_tiles=1 -precompiled
[INFO]  pitonstream,1.0:294: Compiling C test from mem.image for dhrystone.riscv
[INFO]  pitonstream,1.0:300: Creating addr:data map for the test
[INFO]  pitonstream,1.0:268: Found 1 sections
[INFO]  pitonstream,1.0:314: Extracting test sections
[INFO]  make_mem_map.py:387: Checking correctness of section mapping... 
[INFO]  make_mem_map.py:400: Correct!
[INFO]  make_mem_map.py:318: Used 162 out of 33554432 blocks of storage
Loading a test...
100%
TEST OUTPUT >>>

'
The Result is similar to issue 66 (https://github.com/PrincetonUniversity/openpiton/issues/66), but when i compile the same config with 100 MHZ it does work. I tried to halve the UART baud rate but it did not work either. Do you have any idea how to fix this?

Jbalkind, 2021-01-24:
Did you modify piton/tools/src/proto/block.list and the Xilinx IPs (clk_mmcm, UART, and atg_uart_init) to correspond with your frequency change?

RaphaelKlink, 2021-02-01:
Yes, thank you, that fixed the problem.
I had changed the block.list, but not the Xilinx IPs. I had thought, they would be recreated automatically with the new values.

Unfortunately i have a further problem. I try to boot Linux. But the only output i get on the UART is DONE and then nothing else.

I changed the Pull-up Resistors already, programmed the SD Card according to the Instructions and protosyn output lists PITON_FPGA_SD_BOOT.

Could you please help me again.

Jbalkind, 2021-02-01:
You need to flick the pitonstream switch. Whichever one it is that gets rid of the "DONE" print. I haven't used the vcu118 so I don't know exactly which one but you should be able to match the constraints.xdc to the board schematic to see.

Have you read the README entry about the vcu118? Linux support is not all there. Are you using the PMOD SD as mentioned? The board also requires changing frequencies for the SD because of electrical issues that are beyond our control:

```
There is also preliminary support for the VCU118, but not all features work yet on that board. For the VCU118 board you need the PMOD SD adapter from Digilent to be able to use an SD card (the slot on the VCU118 board is not directly connected to the FPGA). As the PMOD0 port has open-drain level-shifters, you also have to replace the R1-R4 and R7-8 resistors with 470 Ohm 0201 SMD resistors on the Digilent PMOD SD adapter to make sure that signal rise times are short enough.
```

RaphaelKlink, 2021-02-02:
What do you mean with the frequency changes?

I lowered the resistors to 100 Ohm, only then the clock was halfway decent (checked with an osciloscop) and it detected the sd card at least sometimes.
 
But it gets stuck during the boot at different stages, it appears the image is not copied correctly to the main memory

Jbalkind, 2021-02-02:
When Michael was setting up the board he reduced the SD clock frequency from the clk_mmcm module to something like 8MHz. I don't remember whether that change is included in the checked-in xci or not