PrincetonUniversity/openpiton #99: cache coherency

jasonzzzzzzz, 2021-03-20:
Hi, I am implement a directory based cache coherency protocol in RTL. I saw OpenPiton github mainpage says the RTL version of cache coherency in OpenPiton is still under developement. I am wondering if there is any updates? or what information/codes I could refer in OpenPiton for my project? Thanks a lot.

Jbalkind, 2021-04-05:
The coherence protocol isn't undergoing any really significant changes. It's been mostly static since the OpenPiton+Ariane release. You can see some documentation on openpiton.org in the microarch spec (though it lacks some details about the RISC-V specific changes). There may still be some other updates that come in future, of course

avinash91231, 2023-08-29:
Hi @jasonzzzzzzz as you have said in above that you have implemented directory based cache coherence protocol in Open piton RTL. So can you please assist me few files so that I could able to figure it out the required files to design Cache coherence Monitor.
Thanks