PrincetonUniversity/openpiton #107: Pitonstream gives timeout when running RISCV benchmarks (Openpiton+Ariane)

aabotaleb, 2022-02-20:
I'm trying to run _OpenPiton+Ariane_
I built it and programmed the FPGA successfully on _Genesys 2.0_ board, also for the _simple hello_world.c_ test it _passes_.
Running **RISCV Benchmark tests** gives timeout.
I used the following command to run the RISCV benchmarks test knowing that the UART is detected at ttyUSB2.
"pitonstream      -b genesys2 -d system -f $PITON_ROOT/piton/design/chip/tile/ariane/ci/riscv-benchmarks.list --core=ariane --precompiled -p ttyUSB2"

(PS: I tried to change "ASM_TIMEOUT_CYCLES " inside "chipset_define.vh" to be 10^11 cycles rather than 5*10^9, but still have timeout after 1500 seconds rather than having it after 75 seconds.)
[image]
I need also to understand the flow of the assembly tests to run the traps , Inspecting "**good_bad_trap_handler.s**", I can see that the written assembly targets **OpenSPARC T1** processor not Ariane core.
What can be the problem and how the trap handling tests are done for the Ariance core?
Thanks.

Jbalkind, 2022-02-20:
Pitonstream requires the test to load a special pass/fail address. The benchmarks that come from riscv-tests don't do this. You might try modifying them to add the load and see whether they work.

https://github.com/PrincetonUniversity/openpiton/blob/openpiton/piton/design/chipset/include/chipset_define.vh#L104-L105

aabotaleb, 2022-02-22:
Thank you a lot for your quick response ^^
I managed partially to solve the problem envisioned by your answer, you can do either of the following:
1- **Modify the makefile** of the riscv-tests added to the open-piton directory by adding -DPITONSTREAM option to the gcc inside the makefile.
This is important to allow sysCall.c functions, specifically "pass" and "fail" access the memory addresses mentioned here: 
https://github.com/PrincetonUniversity/openpiton/blob/openpiton/piton/design/chipset/include/chipset_define.vh#L104-L105
 2- Copy the riscv-tests source code into the ariane folder and add the source code name with the .c extension into the tests.txt.
In this solution, I need to make only single file for each riscv-tests like write the single dhrystone_main.c combines (dhrystone.c and dhrystone_main.c)  files and copy the header file (dhyrstone.h) into "openpiton/piton/verif/diag/assembly/include/riscv/ariane".

In either ways most of the riscv-tests passed.
dhrystone, median, multiply, qsort, rsort, spmv, towers, and vvadd, are executed and exit successfully.

aabotaleb, 2022-02-22:
There's only PMP.c that doesn't pass 
(I think it regards the Physical Memory Protection) tests.
But here it "fails" not "timeout", so I think the current issue is now closed and solved.
I'll create another issue for the PMP problem. 
Thank you a lot.

closed 2022-02-22 by aabotaleb

Jbalkind, 2022-02-22:
There's no need for another issue. The PMPs don't exist in Ariane in the commit that OpenPiton points to. We don't currently plan to use or support them.