PrincetonUniversity/openpiton #109: fatal error: stdint.h: No such file or directory

m-pp, 2022-05-10:
Why do I get this error?

In file included from /home/openpiton/piton/design/chip/tile/ariane/tmp/riscv-tests/build/../benchmarks/common/util.h:41,
                 from /home/openpiton/piton/design/chip/tile/ariane/tmp/riscv-tests/build/../benchmarks/median/median_main.c:12:
/usr/lib/gcc/riscv64-unknown-elf/9.3.0/include/stdint.h:9:16: fatal error: stdint.h: No such file or directory
    9 | # include_next <stdint.h>
      |                ^~~~~~~~~~
compilation terminated.

riscv-gcc toolchain seems to be running fine.

Jbalkind, 2022-05-10:
Doesn't seem to be running fine if you're getting this error. Did you install a system gnu toolchain rather than one built from source? I don't think that debian's riscv64-unknown-elf toolchain works that well, for example. I'd suggest you try grabbing a prebuilt toolchain from elsewhere, like the ones from sifive: https://github.com/sifive/freedom-tools/releases

m-pp, 2022-05-20:
I did that. In addition, there were older versions of gcc that it was picking, which I had to uninstall. That solved this issue.

closed 2022-05-20 by Jbalkind