PrincetonUniversity/openpiton #112: How can we determine SRAM specifications

zlj09, 2022-05-27:
Hi, I am trying to synthesize OpenPiton and perform back-end implementation for a 22nm technology node, following the [manual](http://parallel.princeton.edu/openpiton/docs/synbck_man.pdf). I have a question regarding SRAM integration.

The flow works well until the synthesis stage. I also get some synthesis results without SRAMs. Now I try to integrate SRAMs into the system, but I wonder how to determine the specifications of the SRAMs (depths, bit width, etc.)? I looked into the module setup scripts (${PITON_ROOT}/piton/design/chip/tile/sparc/ffu/synopsys/
script/module_setup.tcl, etc.), but they did not mention the size of the SRAM blocks. 

My understanding is that the SRAM size depends on the cache configurations in design_setup.tcl. But is there a way to determine the required SRAM size? I tried to look into the generated Verilog files, but did not find too many clues. Please let me know if you have any suggestions. Thank you!

Jbalkind, 2022-05-27:
Take a look at the .v.pyv files which produce cache RAMs and see the parameters that are being used in those. Try searching the repo on github for `MakeGenericCacheDefine` to see those instances

jimaandro, 2024-04-08:
> Take a look at the .v.pyv files which produce cache RAMs and see the parameters that are being used in those. Try searching the repo on github for `MakeGenericCacheDefine` to see those instances

Hello, I have a similar issue. Actually I want to replace the ddr with a custom memory controller. My controller has 64 bit data width AXI interface, so I have to use the noc_to_axi4. But I understand that this is customized for ddr4 AXI with 512 bits of data
Is there any way to customize the noc path to ddr to send 64 bits?
I can't customize only the noc to axi4 because this will not give valid every 64 bits

Jbalkind, 2024-04-08:
I'd suggest you could try using an AXI downsizer like the one provided by PULP (or a Xilinx IP) https://github.com/pulp-platform/axi/blob/master/src/axi_dw_downsizer.sv

jimaandro, 2024-04-08:
ok thank you very much! I will try this