PrincetonUniversity/openpiton #115: Noc output data path with chip_bridge

Kit-lewiz, 2022-06-29:
In the chipset.v file, there's a set of noc_data/noc_valid I/O's for no chip_bridge. 

Is there a way to use use those with chip_bridge? I want to use those noc_data/valid as inputs to my own module. I can see that it goes through credit_to_valrdy, chipset_imp1, and fpga_bridge which is kind of confusing. can someone help with the noc_data path?

Jbalkind, 2022-06-30:
I'm not quite sure what you're asking for. The chip bridge is turning three 64 bit networks into a single 32 bit link. In the NO_CHIP_BRIDGE case, those three links are directly wired off the chip. We have I/O modules for when chip_bridge is in use, not for NO_CHIP_BRIDGE.

Could you be a little more specific about what you're looking for?

Kit-lewiz, 2022-06-30:
[image]

Referring to the image attached. I want to add a new core that takes the 64 bit noc signals and is connected through the chipset. But I want to bypass all the xbar, bootrom and other stuff that's part of the chipset. What configuration should I be using here? And is it possible to bypass the BOOTROM and other UART components?

Jbalkind, 2022-07-01:
When you say "core", what exactly do you mean? It doesn't make sense to "bypass" the crossbar etc if you want that core to talk to other devices/cores. You can probably just connect whatever the device is to the chipset crossbars. Perhaps you could send me an email with more detail if you don't want to give it in a public forum?

Kit-lewiz, 2022-07-05:
Sure, I sent you an email with more detail.