PrincetonUniversity/openpiton #116: Random Output (Raw Binary) data when running Openpiton+Ariane on VCU118

aabotaleb, 2022-09-08:
I generated a new project for openpiton+ariane with the VCU118 is the target board using the following command:

protosyn -b vcu118 -d system --core=ariane --uart-dmw ddr --x_tiles=2 --y_tiles=2

Then when trying to run any c source code via UART , I got random binary data printed on the terminal.

Are there any recommendations about what to do or modify inside the RTL to solve this problem.
PS: I'm using the openpiton-dev branch from openpiton repo.  
Thank you in advance. 

[image]

Jbalkind, 2022-09-12:
I have a hunch that this is due to the transitioning of pitonstream to python3. We didn't test everything thoroughly and so it may just be printing the string in a way that looks like garbage. Could you try reverting pitonstream (and its dependencies?) specifically back to the older python2 versions and try again? It would be good to track down exactly what we need to fix in the uart->output printing path for python3.

aabotaleb, 2022-10-28:
Thank you so much Professor. 
It was a while since I was working with VCU118 and
side note , that sometimes a totally wrong binary output appears in terminal when working with wrong USB port.
i.e. you need to specify correct port as "pitonstream" option "-p ttyUSB1" rather than default port for example.

closed 2022-10-28 by aabotaleb

Jbalkind, 2024-03-20:
I believe (as noted above) that they were opening the wrong tty when running pitonstream?

khoatm98, 2024-03-21:
Thanks @Jbalkind , i just run it on vcu118 and it works now