PrincetonUniversity/openpiton #117: Boot linux on xilinx vcu108 board

allon-nolla, 2022-09-19:
Hi everyone,I'm trying to boot linux on vcu108 dev board,The 108 is almost the same as 118.But the system stuck at "copying block 0 of 1 blocks (0 %)"
These are something I have already done for transplanting from the 118project to 108project.
1. Generated a vcu118 FPGA proto project(ariane core).
2. Copied all the the files into vivado project.
3. Modified board's type,upgrade IPs 
    (ddr4 mig IP is using a 300mhz clk).
5. Modified the IO ports constraints.
6. Deleted the (*IOB = TRUE*)setting in "sd_cmd_serial_host.v" and "sd_data_serial_host.v" because of some errors.
7. Did synthesis,implementation,and bitstream generated.
8. Connected a 3rd party SD slot with several changed 470ohm resistance,and prepared a SD card.
   (Connection between dev board PMOD and SD slot's interface is some jumper wires,does it influence the system?)
10. Downloaded bit file into FPGA and see what happened in the UART screen.

Problems:
1. How can I lower the frequency?
For example,If I want to use a 80Mhz clk rather than 100Mhz,I need to change the mmcm IP setting,the mmcm_chip IP setting.And for UART baud rate115200,number(in hex) in the "uart_data.coe" is also needed to be changed.
But after that, what I can see via UART is nothing but messy code.
2.  Both vcu118 and vcu108 have a mcrio SD slot which is not connected to the FPGA package pin,so we need a SD slot.But my system is stucking at "copying block 0 of 1 blocks (0 %)",It seems like the system can not get anything from the SD card.

It would be appreciated that any suggestion about solving problems.
Thanks :)

Jbalkind, 2022-09-19:
When you lower the frequency, you need to also change the frequency for your board in block.list as this gets passed to the bootrom generation (and the bootrom needs to know the frequency to correctly calculate the baud rate). If you fix this then you should stop seeing garbage. https://github.com/PrincetonUniversity/openpiton/blob/openpiton/piton/tools/src/proto/block.list#L28

If you have another supported board available, I'd suggest you try with that first because some SD cards are somehow non-compliant and don't work with the controller. You could also just try a different SD or two.

allon-nolla, 2022-09-27:
> When you lower the frequency, you need to also change the frequency for your board in block.list as this gets passed to the bootrom generation (and the bootrom needs to know the frequency to [...]

Thank you very much and sorry to reply so late.As you told me,the chipset clk is 50Mhz now and slack looks better.But the system still stuck as before.We do have a sd card working on genesys2,which still can not work on vcu108.
And about the change of frequency,is there any modifications that i need to do for CDC stuff?

Jbalkind, 2022-09-28:
Can you try using pitonstream (add `--uart-dmw ddr` to protosyn and flip the appropriate DIP switch)? That would let you check if the design is otherwise working and potentially narrow things down.
My recollection is that you don't need to make other CDC related changes.