PrincetonUniversity/openpiton #121: Vivado Project was not created properly

suppamax, 2022-12-30:
When I try to synthesize CVA6 on a Genesys2 board with Vivado 2022.2 I get the following 
```
[max@yarra build]$ protosyn -b genesys2 -d system --core=ariane --uart-dmw ddr
[INFO]  protosyn,2.5:702: ----- System Configuration -----
[INFO]  protosyn,2.5:720: x_tiles   = 1
[INFO]  protosyn,2.5:721: y_tiles   = 1
[INFO]  protosyn,2.5:722: num_tiles = 1
[INFO]  protosyn,2.5:729: core      = ariane
[INFO]  protosyn,2.5:732: defining RTL_TILE0
[INFO]  protosyn,2.5:762: setenv RTL_ARIANE0
[INFO]  protosyn,2.5:780: network   = 2dmesh_config
[INFO]  protosyn,2.5:784: l15 size  = 8192
[INFO]  protosyn,2.5:785: l15 assoc = 4
[INFO]  protosyn,2.5:786: l1d size  = 8192
[INFO]  protosyn,2.5:787: l1d assoc = 4
[INFO]  protosyn,2.5:788: l1i size  = 16384
[INFO]  protosyn,2.5:789: l1i assoc = 4
[INFO]  protosyn,2.5:790: l2  size  = 65536
[INFO]  protosyn,2.5:791: l2  assoc = 4
[INFO]  protosyn,2.5:805: ---- Additional RTL Defines ----
[INFO]  protosyn,2.5:808: NO_RTL_CSM
[INFO]  protosyn,2.5:808: PITON_FPGA_MC_DDR3
[INFO]  protosyn,2.5:808: PITONSYS_MEM_ZEROER
[INFO]  protosyn,2.5:808: PITON_FPGA_SD_BOOT
[INFO]  protosyn,2.5:808: PITONSYS_UART_BOOT
[INFO]  protosyn,2.5:808: PITON_NO_CHIP_BRIDGE
[INFO]  protosyn,2.5:808: PITON_UART16550
[INFO]  protosyn,2.5:808: PITON_FPGA_ETHERNETLITE
[INFO]  protosyn,2.5:810: --------------------------------
[INFO]  protosyn,2.5:879: Generating UART init sequence
[INFO]  protosyn,2.5:631: Using core clock frequency: 66.667 MHz
[INFO]  protosyn,2.5:285: Building a project for design 'system' on board 'genesys2'
[INFO]  protosyn,2.5:330: Running FPGA implementation down to bitstream generation
[INFO]  protosyn,2.5:932: Checking Project Build results
[ERROR] fpga_lib.py:344: Vivado Project was not created properly!
[ERROR] fpga_lib.py:345: Check: /home/max/Workarea/openpiton/build/genesys2/system/protosyn_logs/make_project.log
```
The tail of make_project.log is 
```
INFO: compiling DTS and bootroms for Ariane (MAX_HARTS=1, UART_FREQ=66667000)...
dtc -I dts ariane.dts -O dtb -o ariane.dtb
riscv64-unknown-elf-gcc -Tlinker.ld bootrom.S -nostdlib -static -Wl,--no-gc-sections -o bootrom.elf
riscv64-unknown-elf-objcopy -O binary bootrom.elf bootrom.bin
dd if=bootrom.bin of=bootrom.img bs=128
python ./gen_rom.py bootrom.img
rm bootrom.bin bootrom.elf ariane.dtb
child process exited abnormally
    while executing
"exec make all 2> /dev/null"
    invoked from within
"if  {[info exists ::env(PITON_ARIANE)]} {
  puts "INFO: compiling DTS and bootroms for Ariane (MAX_HARTS=$::env(PITON_NUM_TILES), UART_FREQ=$env(CONFI..."
    (file "/home/max/Workarea/openpiton/piton/tools/src/proto/common/setup.tcl" line 121)

    while executing
"source $DV_ROOT/tools/src/proto/common/setup.tcl"
    (file "/home/max/Workarea/openpiton/piton/tools/src/proto/vivado/setup.tcl" line 30)

    while executing
"source $DV_ROOT/tools/src/proto/vivado/setup.tcl"
    (file "/home/max/Workarea/openpiton/piton/tools/src/proto/vivado/gen_project.tcl" line 33)
INFO: [Common 17-206] Exiting Vivado at Fri Dec 30 16:42:08 2022...
```
In my understanding the failing `make` is the one executed under `bootrom/baremetal`, but if I run it manually I see no problem.
Any suggestion?
Thanks

suppamax, 2023-01-10:
Using the project under `openpiton-dev` branch has not such issue

closed 2023-01-10 by suppamax

yelabWang, 2023-06-02:
May I ask if the problem has been resolved? I am currently encountering the same problem, but using the dev branch still results in the same error.

Jbalkind, 2023-06-02:
Perhaps try reinitialising your ariane submodule?

yelabWang, 2023-06-02:
> Perhaps try reinitialising your ariane submodule?

I have tried, but it seems to have no effect. The project has been regenerated several times.

Jbalkind, 2023-06-02:
Please post the command you're running and the output you are seeing.

yelabWang, 2023-06-02:
> Please post the command you're running and the output you are seeing.

[INFO]  protosyn,2.5:705: ----- System Configuration -----
[INFO]  protosyn,2.5:723: x_tiles   = 1
[INFO]  protosyn,2.5:724: y_tiles   = 1
[INFO]  protosyn,2.5:725: num_tiles = 1
[INFO]  protosyn,2.5:732: core      = ariane
[INFO]  protosyn,2.5:735: defining RTL_TILE0
[INFO]  protosyn,2.5:766: setenv RTL_ARIANE0
[INFO]  protosyn,2.5:784: network   = 2dmesh_config
[INFO]  protosyn,2.5:788: l15 size  = 8192
[INFO]  protosyn,2.5:789: l15 assoc = 4
[INFO]  protosyn,2.5:790: l1d size  = 8192
[INFO]  protosyn,2.5:791: l1d assoc = 4
[INFO]  protosyn,2.5:792: l1i size  = 16384
[INFO]  protosyn,2.5:793: l1i assoc = 4
[INFO]  protosyn,2.5:794: l2  size  = 65536
[INFO]  protosyn,2.5:795: l2  assoc = 4
[INFO]  protosyn,2.5:814: ---- Additional RTL Defines ----
[INFO]  protosyn,2.5:817: NO_RTL_CSM
[INFO]  protosyn,2.5:817: PITON_FPGA_MC_DDR3
[INFO]  protosyn,2.5:817: PITONSYS_MEM_ZEROER
[INFO]  protosyn,2.5:817: PITON_FPGA_SD_BOOT
[INFO]  protosyn,2.5:817: PITONSYS_UART_BOOT
[INFO]  protosyn,2.5:817: PITON_NO_CHIP_BRIDGE
[INFO]  protosyn,2.5:817: PITON_UART16550
[INFO]  protosyn,2.5:817: PITON_FPGA_ETHERNETLITE
[INFO]  protosyn,2.5:819: --------------------------------
[INFO]  protosyn,2.5:888: Generating UART init sequence
[INFO]  protosyn,2.5:634: Using core clock frequency: 66.667 MHz
[INFO]  protosyn,2.5:287: Building a project for design 'system' on board 'genesys2'
[INFO]  protosyn,2.5:332: Running FPGA implementation down to bitstream generation
[INFO]  protosyn,2.5:941: Checking Project Build results
[ERROR] fpga_lib.py:344: Vivado Project was not created properly!
[ERROR] fpga_lib.py:345: Check: /home/xxx/openpiton/build/genesys2/system/protosyn_logs/make_project.log

in the make_project.log

Info: /home/xxx/openpiton/piton/design/common/rtl/bram_sdp_wrapper.v.pyv exists! Preprocessing...
no such variable
    (read trace on "::env(PYTHONHOME)")
    invoked from within
"set tmp_PYTHONHOME $::env(PYTHONHOME)"
    invoked from within
"if {[file exists ${PYV_IMPL_FILE}]} {
            puts "Info: ${PYV_IMPL_FILE} exists! Preprocessing..."
    
            # Setup temporary filename f..."
    ("foreach" body line 4)
    invoked from within
"foreach RTL_IMPL_FILE ${RTL_IMPL_FILES} {
        set PYV_IMPL_FILE "${RTL_IMPL_FILE}.pyv"
        # File gets preprocessed if a PYV version exists
  ..."
    (procedure "pyhp_preprocess" line 3)
    invoked from within
"pyhp_preprocess ${ALL_RTL_IMPL_FILES}"
    invoked from within
"set ALL_RTL_IMPL_FILES [pyhp_preprocess ${ALL_RTL_IMPL_FILES}]"
    (file "/home/xxx/openpiton/piton/tools/src/proto/common/setup.tcl" line 119)

    while executing
"source $DV_ROOT/tools/src/proto/common/setup.tcl"
    (file "/home/xxx/openpiton/piton/tools/src/proto/vivado/setup.tcl" line 30)

    while executing
"source $DV_ROOT/tools/src/proto/vivado/setup.tcl"
    (file "/home/xxx/openpiton/piton/tools/src/proto/vivado/impl_flow.tcl" line 33)
INFO: [Common 17-206] Exiting Vivado at Sat Jun  3 00:40:03 2023...

Jbalkind, 2023-06-02:
You're seeing a completely different error.

Somehow you have python installed but your PYTHONHOME environment variable is not set. Perhaps just `export PYTHONHOME=""` or something along those lines? Alternatively you could potentially comment out these two blocks:

https://github.com/PrincetonUniversity/openpiton/blob/openpiton-dev/piton/tools/src/proto/common/setup.tcl#L127-L132
https://github.com/PrincetonUniversity/openpiton/blob/openpiton-dev/piton/tools/src/proto/common/setup.tcl#L155-L156

yelabWang, 2023-06-02:
Potentially comment  doesn't work, it's the same error. How do these two environment variables need to be set?

Jbalkind, 2023-06-02:
They don't need to be set to a particular value. If you read the six lines of tcl, you can see that it's saving the old value of both PYTHONHOME and PYTHONPATH and restoring them afterwards. That's because the python packaged with vivado almost always conflicts with the system python.

Are both your PYTHONHOME and PYTHONPATH unset before you call protosyn? Perhaps only one of them is set and you need to just comment one of the two. Seeing the output would be helpful to diagnose

yelabWang, 2023-06-02:
Yes, I unset two environment variables, but there is still the same problem

yelabWang, 2023-06-02:
[image]

[image]

Looks like the same problem as before

Jbalkind, 2023-06-02:
You say you commented the code out. How could the error occur on the same line if it's been commented out? Either comment it out, or set the variables to "" and see if that works. Do only one or other, not both. Send the output that you see.

yelabWang, 2023-06-02:
In fact, I have tried both methods separately, but the result is the same, still the problem. The output from the command line is the same.

Jbalkind, 2023-06-02:
Are you sure you're running the same code that you're modifying? Sometimes people have multiple copies of the repo and don't realise they're modifying different code from what they're running. If you commented out a line then it can't cause an error.

yelabWang, 2023-06-02:
Yes, I confirm that I do not have any other copies

yelabWang, 2023-06-02:
I solved it. The file openpiton/piton/tools/src/proto/common/pyhp_preprocess.tcl about PYTHONPATH PYTHONHOME need to be commented out. Thank you very much!

grigoriy-chirkov, 2023-06-02:
@Jbalkind tbf, I've had this issue with missing PYTHONHOME and PYTHONPATH on all of the systems where I attempted to synthesize OpenPiton (even though I obviously have python installed everywhere). So, the assumption that everybody has these variables set is not the best one

Jbalkind, 2023-06-02:
It's odd because I've never had the issue. Perhaps someone could just add a check for each variable before saving them?