PrincetonUniversity/openpiton #122: OpenOCD error: Unknown RTOS type riscv

suppamax, 2023-01-12:
I've created a system with 2 CVA6 cores.
The implementation on Genesys2 is succesful.
I've tried connecting openOCD by using the `ariane-multi-hart.cfg` configuration file, but I get
```
../ariane-multi-hart.cfg:27: Error: Unknown RTOS type riscv, try one of: ThreadX, FreeRTOS, eCos, linux, chibios, Chromium-EC, embKernel, mqx, uCOS-III, nuttx, RIOT, Zephyr, hwthread,  or auto
```

Can you suggest how to modify the .cfg file or which version of openOCD to use?

tianrui-wei, 2023-01-16:
Here's the openocd configuration I use for using with BSCANE2
```
adapter driver ftdi
adapter speed 100
transport select jtag

# From Digilent support:
# The SRST pin is [...] 0x20 and 0x10 is the /OE (active low output enable)

ftdi_device_desc "Digilent Adept USB Device"
ftdi_vid_pid 0x0403 0x6010
# channel 1 does not have any functionality
ftdi_channel 0
# just TCK TDI TDO TMS, no reset
#
ftdi_layout_init 0x0088 0x008b
reset_config none

#set _CHIPNAME riscv
#jtag newtap fpga_unused tap -irlen 6 -expected-id 0x33651093
#jtag newtap $_CHIPNAME cpu -irlen 6 -expected-id 0x14B79093
#
#set _TARGETNAME $_CHIPNAME.cpu
#target create $_TARGETNAME riscv -chain-position $_TARGETNAME
#
#
#riscv set_ir idcode 0x9249
#riscv set_ir dtmcs 0x22924
#riscv set_ir dmi 0x23924

set _CHIPNAME riscv
jtag newtap $_CHIPNAME cpu -irlen 6 -expected-id 0x33687093

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME riscv -chain-position $_TARGETNAME

riscv set_ir idcode 0x9
riscv set_ir dtmcs 0x22
riscv set_ir dmi 0x23
```

We're currently using openocd 0.11. Let me know if you run into any issues.

suppamax, 2023-01-24:
Hi Tianrui,

thanks for your reply.
Let me summarize the status here.

1. I generated a dual core Ariane processor for Genesys2 with
`protosyn -b genesys2 -d system --x_tiles=2 --core=ariane --uart-dmw ddr`
2. I uploaded the bitfile
I see the "OpenPiton..." message being displayed on the LCD screen on the board
3. I tried to connect with openocd, using your config file
```
Open On-Chip Debugger 0.11.0-rc1+dev (SiFive OpenOCD 0.10.0-2020.12.1)
Licensed under GNU GPL v2
For bug reports:
	https://github.com/sifive/freedom-tools/issues
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 100 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x00000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x0)
Warn : JTAG tap: riscv.cpu       UNEXPECTED: 0x00000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x0)
Error: JTAG tap: riscv.cpu  expected 1 of 1: 0x33687093 (mfg: 0x049 (Xilinx), part: 0x3687, ver: 0x3)
Error: Trying to use configured scan chain anyway...
Warn : Bypassing JTAG setup events due to errors
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
Warn : target riscv.cpu examination failed
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
```
which looks wrong to me.

I connect to the board via the microUSB connector labelled `JTAG`. Is that correct?

tianrui-wei, 2023-01-25:
Hi Max,

This issue is perhaps easier to discuss offline. Can we continue this in openhwgroup Mattermost?

Thanks,
Tianrui

closed 2023-02-06 by suppamax