PrincetonUniversity/openpiton #124: Ariane cache operations

TallAllie, 2023-02-14:
Esteemed Colleagues!

I have three unrelated questions...  Please forgive (and correct) me if this is not the best place to ask them...  :-)

1.) How can an Ariane tile flush a write all the way to DDR?

- Specifically, suppose an Ariane core wants to write a byte at address `0x8001_4200`.
- To BYPASS the L1 and L1.5 cache, it can ACTUALLY create an IO write by using physical address `0x80_8001_4200`  (setting bit #39).
- With that, the write data leaves the tile and sits dirty in the L2.
- Next, as described in the microarchitecture specification for the L2, the Ariane core must do a specific `ldub` instruction to flush the cacheline from the L2 to DDR.
- To compose that `ldub` instruction, it must know the index/way/tag/etc. information about where the line is in the L2.
- My question now becomes ... what is the recipe to map a physical address to this special `ldub` instruction?  How can the code running on the Ariane create the correct instruction?
- Obviously, if I reconfigure the attributes of the L2, that mapping changes.  How?

    1.5) If we turn on page tables, marking the page containing `0x8001_4200` as uncacheable, would the L2 respect that?

2.) How can an Ariane core discard a line from the L1, L1.5, and L2?

- Suppose an Ariane core is polling a location in DDR, but suppose something in the simulation infrastructure magically reaches into DDR and changes the byte being polled.
- How can the Ariane see this new value?
- Typically, we issue an invalidate cacheop instruction that asks the hardware to simply discard a clean line, forcing the next read to fetch from DDR.

  2.5) Again if we turned on page tables, and marked the page containing the polled location as uncacheable, would the L2 respect that?

3.) I want to add another small peripheral to the `riscv_peripherals.sv` module (and to the memory map).  Can you point me to an example/recipe where someone has done something similar?

Thank you!

Jbalkind, 2023-02-28:
Ok there's a lot to this so I haven't been able to budget the time to answer it all before now. In general, it's easier to find time to answer smaller questions.

1. Your understanding here isn't quite right. `0x8001_4200` and `0x80_8001_4200` are completely different addresses. Setting the 1 bit does not do a NC access to an otherwise cacheable address. The core must use the NC message types to the L1.5 if it wants to do such requests. The 1 bit in bit 39 was simply the behaviour that SPARC used for its IO space.

Regarding building the address for a flush: The flush you looked at is described in the microarchitecture spec as a coherence flush. What you actually want is the displacement flush (sec 5.4.11 in http://openpiton.org/docs/micro_arch.pdf) which reorders the address bits but then does not require index and way to flush the line.

1.5. That depends on your core. CVA6 at present does not have this capability (RISC-V only standardised this recently). The OST1 core does. Provided the correct message types enter the L1.5, it should behave correctly.

2. We don't have that feature right now in P-Mesh but we would really like to support it. If you're willing to help with this then I'd be happy to discuss the details. Our colleagues at BSC have been asking about this too.

2.5. See 1.5. Easier is to just allocate the memory region as non-cacheable from the beginning.

3. I'm not sure why your peripheral needs to be in riscv_peripherals.sv? That module itself is a clear example of what you need to do. See the wiring it entails in chipset_impl.v.pyv plus the devices_ariane.xml files where the entries are added.

TallAllie, 2023-03-04:
Sorry @Jbalkind!  I appreciate your response and corrections, and especially for your TIME!  :-)

Yes, I was confused about how the features of different cores were leveraged in OpenPiton. I have non-cacheable reads and writes working now, but I think you're right that the best thing to do (with CVA6) would be to enable the MMU and configure non-cacheable pages where I need (covering #1, #1.5, #2.5, and #3).

For #1.5, actually, I do want both a coherence flush (where dirty data is written to memory [as with an eviction]), and a displacement flush (where clean data is just discarded).

Perhaps you can point me to some C or ASM code that performs these kinds of operations?  My hope is to understand how to implement something like `flush_magic()` and `evict_magic()` below.  Given an address, what is the recipe to perform the cacheOps described in your microarchitecture specification?

```
#include <stdlib.h>
#include <stdint.h>

extern uint64_t* flush_magic(uint64_t* addr);
extern uint64_t* evict_magic(uint64_t* addr);

void foo(void) {
    // Here we WRITE to an address.
    // Then we FLUSH the write data to DDR.
    uint64_t *wr_addr = malloc(sizeof(uint64_t));
    uint64_t *flush = flush_magic(wr_addr);
    
    *wr_addr = 0x9876543210fedcba;  // write
    asm volatile (
        "fence   rw, rw                  \n\t"
        "ldub    t1, 0(%[flush])         \n\t"  // OpenPiton flush
        : : [flush] "A" (flush)
        : "t0", "memory"
    );

    // Here we continually READ/POLL an address until the
    // value in memory changes.
    // To do this, we evict earlier reads from the cache.
    uint64_t *rd_addr = malloc(sizeof(uint64_t));
    uint64_t *evict = evict_magic(rd_addr);

    while (*rd_addr == 0x0ULL) {
        asm volatile (
            "fence   rw, rw         \n\t"
            "ldub  t0, 0(%[evict])  \n\t"  // OpenPiton evict
            : : [evict] "A" (evict)
            : "t0", "memory"
        );
    }
}
```

Jbalkind, 2023-03-04:
Sorry, no need to apologise. I'm just trying to be clear about my user interface.

Let me be a little clearer here: I don't think that coherence and displacement flushes behave differently. Based on this code segment (https://github.com/PrincetonUniversity/openpiton/blob/openpiton-dev/piton/design/chip/tile/l2/rtl/l2_pipe1_ctrl.v.pyv#L3110 and line 3122) I'm pretty sure that both types of flush perform a writeback to memory if the data in the L2 ends up dirty after L1.5 invalidations. That's the same as the first case you ask for but not the second. The difference between the two should only be whether they use the address or the index+way to choose the line to flush.

I think that we could add a pair of new message types (say MSG_TYPE_L2_DIS_INVAL_REQ and MSG_TYPE_L2_LINE_INVAL_REQ? That would correspond more directly with the names used in the RISC-V CMO specification https://github.com/riscv/riscv-CMOs/blob/master/specifications/cmobase-v1.0.1.pdf) that drop the data even when it is dirty. I believe that would mainly just entail making some modifications in l2_pipe1_ctrl though it may extend a bit further than that.

Again, regarding the MMU - Ariane does not currently have the ability to indicate cacheability through its page table entries. The MMU would need to be upgraded to support the RISC-V Svpbmt extension. Once that's done, we would still need to do some testing to be sure that it behaved correctly because we've so far assumed that Ariane cannot change between cacheability modes for a fixed address (unlike the SPARC core).

Here's a snippet from Marcelo which should build the necessary address needed to do the flush:
`addr = ((uint64_t)data_paddr & 0x03ffffffc0) | 0xac00000000 | ((uint64_t)data_paddr >> 34);`

TallAllie, 2023-03-07:
Thanks again @Jbalkind (and Marcello)!

To summarize...

- What I call a "flush" (using Marcelo's math) works.  Dirty data in the caches is written to memory.

- What I call a "discard"/"invalidate", is not supported today.  Clean data from an earlier load remains in the caches, and of course any new value that magically appears in DDR is never seen (without forcing a natural eviction of that line, which is untenable).

I have not attempted to do anything with the MMU, custom paging, pmp, etc.  This is next.  :-)

closed 2023-03-07 by TallAllie

Jbalkind, 2023-03-07:
Great - if you would like to chat as you go then please consider also joining our OpenPiton Zulip (openpiton.zulipchat.com)