PrincetonUniversity/openpiton #125: Debian GUI and networking capabilities

0ena, 2023-03-03:
Hi,

I see that the OpenPiton version of the CVA6 can run full-stack Debian Linux.
I have the following two questions if you can kindly please answer:

1) does your implementation support a Linux GUI? For example the Genesys 2 has a VGA port on it, can this be used with your implementation to run Debian with GUI?

2) does your implementation support networking? For example can the ethernet port on the Genesys 2 board allow OpenPiton to get an IP and connect to the internet, support packet exchange, etc.?

Thank you in advance for your response and time!

Kind regards,
Nassos

Jbalkind, 2023-03-05:
Hi Nassos!

Ethernet works and we've done tutorials where attendees can connect to the internet and use e.g. lynx as a terminal-based web browser. We have also used graphical apps over X forwarding. We have some patches internally to support the VGA and PS2 on the Genesys 2 which worked with the OST1 but we simply haven't taken the time to test those with CVA6 yet. I know that the old lowRISC SoC was demoed with CVA6 running a GUI with a similar setup so I think it would just be a matter of resolving some hardware and software plumbing issues.

Thanks,
Jon

0ena, 2023-03-28:
I appreciate your response Jon!

closed 2023-03-28 by 0ena