PrincetonUniversity/openpiton #128: Question about run personal program on Araine genesys2

yelabWang, 2023-06-03:
I tried run the personal code on genesys2, but when I use the function malloc, it broke. Then I changed the rv64_cc file to enable the stdlib,but it still not works.  I want to know whether the program support the stdlib?

[image]

[image]

yelabWang, 2023-06-03:
This problem mainly occurs when the function calls malloc, and running malloc and free in the main function is not a problem.

yelabWang, 2023-06-03:
When the space of malloc increases, an error will also be reported

Jbalkind, 2023-06-07:
We don't support the use of malloc in our bare metal environment. If you're bringing your own binaries that you're building yourself then it's your responsibility to handle providing a functioning bare metal libc.