PrincetonUniversity/openpiton #129: Q: why are we using 2 rb + 4 lru bits to implement pseudo LRU algorithum?

Nameer-Iqbal-Ansari, 2023-06-05:
In the pseudo LRU algorithm we can maintain a track for lru way with the 3 bits tree based pseudo lru algorithm, but we are using 4 bits for lru and 2 rb bits as to start searching for the possible replacement way.

fei-g, 2023-06-08:
There are different implementation choices when doing a pseudo LRU. I agree that our current implementation is not the most resource-efficient one in terms of the number of bits used. 

Students have characterized different cache replacement policies in OpenPiton as course projects. But I don't remember if anyone compared the one-bit-per-way PLRU with the tree-based PLRU. You are welcome to implement and characterize that.

Nameer-Iqbal-Ansari, 2023-06-09:
HI fei-g,
Yes, Since me and my group members are doing our graduation Final year project on OpenPiton we found out about this, So yes we can work on this specific.