PrincetonUniversity/openpiton #131: midas: FATAL ERROR: Command "pal diag.pal  > build/diag.src" failed with status 2.

qazihamidullah, 2023-07-31:
In the step Running a Regression, after writing the command sims **-sim_type=vcs -group=tile_mini** in the setup phase we get the following errors 
midas: ##  SETUP PHASE
midas: ###########################################################
midas: ### Will build in directory "/var/tmp/qazi:tr_intr00:default:tile1_mini:0:11964:job92190540/build"
midas: mkdir /var/tmp/qazi:tr_intr00:default:tile1_mini:0:11964:job92190540/build
midas: pal diag.pal  > build/diag.src
Can't write '/usr/tmp/diag.pal.12085.pl' No such file or directory
midas: At pkg=Midas::Interface, file=/home/qazi/Documents/OpenPiton/piton/tools/perlmod/Midas/3.30/lib/site_perl/5.8.0/Midas/Interface.pm, line=370
midas: FATAL ERROR: M_CMDFAIL (#5): Command failed.
midas: FATAL ERROR: Command "pal diag.pal  > build/diag.src" failed with status 2. 
sims: Caught a SIGDIE. midas compilation error at /home/qazi/Documents/OpenPiton/piton/tools/src/sims/sims,2.0 line 2400.

Jbalkind, 2023-07-31:
Hm I think maybe you need to set DRMJOBSCRATCHSPACE to a directory you have permissions on. Could you give that a try?

qazihamidullah, 2023-08-02:
Hello! Thanks for the reply.  
I assigned a new directory for DRMJOBSCRATCHSPACE in the bashrc file but it is still giving the same error as above.
After reading a documentation about DRMJOBSCRATCHSPACE, I think it creates scratch space for the regression and running simulation etc for vcs simulator only but I am using Questasim. Is there any other way through which I can create scratch space which Questasim can access.

Jbalkind, 2023-08-02:
Ah the error is with midas trying to write to /usr/tmp, not /var/tmp

Try editing here? https://github.com/PrincetonUniversity/openpiton/blob/3cc7bf4d3d1ee2f8e18c33eda6c136a57222806b/piton/tools/local/pal/1.13/bin/pal#L209-L210

qazihamidullah, 2023-08-08:
The error is solved now. Thanks for the support and guidance.

closed 2023-08-08 by qazihamidullah