PrincetonUniversity/openpiton #139: Are There Existing Implementations for Assigning Specific Computations to Specific Processor Cores in OpenPiton and Linux?

pychen121, 2023-11-22:
Hi,

I am working on a project that utilizes the OpenPiton framework and boots a Linux operating system. I seek to assign specific computations to designated processor cores. Has anyone attempted such an implementation? Alternatively, how can I modify the operating system scheduling to accomplish this objective?

Jbalkind, 2023-11-22:
For our prior work we have simply used `taskset` which is a linux utility to put applications on specific cores.

pychen121, 2023-11-24:
@Jbalkind Thank you for your help. I will definitely try your suggestion. 
Is it possible to set the direction of routers of a tile so that the output of a tile is always directed to an adjacent tile? If so, what files can I refer to and configure?

Jbalkind, 2023-11-26:
This isn't really how openpiton works. The L2 cache is distributed across all tiles. Traffic depends on which parts of the address space are being accessed. Requests are routed via multiple tiles to get to their destinations.

pychen121, 2023-11-29:
@Jbalkind Thank you for your help. If I want to specify that the result of a code executed in Tile1 is stored at a specific address in the L2 cache, and Tile2 accesses this address to get the data for subsequent calculations, what files do I need to modify to achieve this effect?

Jbalkind, 2023-12-06:
If you're running bare metal and you're writing C, then you can use whatever addresses you like in your C code. You don't need to modify the OpenPiton design to make that possible. You could check (and potentially change) which home allocation method (https://github.com/PrincetonUniversity/openpiton/blob/3cc7bf4d3d1ee2f8e18c33eda6c136a57222806b/piton/design/chip/tile/rtl/config_regs.v.pyv#L180) you use to make the job easier.

pychen121, 2023-12-08:
@Jbalkind Thank you very much for your suggestions. I will try it.