PrincetonUniversity/openpiton #141: Baremetal booting

jimaandro, 2024-01-13:
Hi
I am trying to run a baremetal C program (I have converted it to bin and then I transfered it to DDR at 0x80000000). I want to use the baremetal bootloader. How should I configure the HW in order to boot in baremetal mode?

Jbalkind, 2024-01-15:
I'd need a little more information to understand what you're describing - Is this on FPGA? In simulation? How did you transfer the binary to DDR?

Jbalkind, 2024-01-16:
Ah ok got it. You have two options. One is you can just change the sense of the instantiation of the bootrom inside riscv_peripherals.sv so that the bare metal one is instantiated instead of the linux one. The other is you can modify the linux bootrom to remove the call to `sd_copy()`. The latter is what our colleagues at BSC do and it seems to work best for them.

closed 2024-02-21 by jimaandro