PrincetonUniversity/openpiton #149: Riscv qemu missing

ChiaLiu0618, 2024-07-10:
Hi, 
I am a student working under professor Kun-Chih(Jimmy) Chen.
We are currently working on a project that would use OpenPiton.
When we were building the files, it seems that the link doesn’t work when it tried cloning riscv-qemu from a GitHub link. 
The files from the link seems to have been deleted by the author.
Here is our presentation containing information about this problem.

Thank you～
[NoC_0710.pptx](https://github.com/user-attachments/files/16157114/NoC_0710.pptx)

ChiaLiu0618, 2024-07-25:
I have been using the RISC-V toolchain from this repository "https://github.com/riscv-collab/riscv-gnu-toolchain.git". The build was successful, but I encountered the following error when I tried to build and run the model. Could you please provide any suggestions on what might be causing this issue?
[image]

Jbalkind, 2024-07-25:
Which version of verilator are you using? I would suggest you update to the latest release from github and then try (though note you will have to add -vlt_build_args=--no-timing to your sims build command. You should also delete your build/manycore/rel-0.1/obj_dir/ to avoid any other build weirdness)

Jbalkind, 2024-08-12:
Could you show me your sims commands?

ChiaLiu0618, 2024-08-12:
Hi, these are the commands I used.
sims -sys=manycore -x_tiles=2 -y_tiles=2 -vlt_build -ariane -vlt_build_args=--no-timing
sims -sys=manycore -vlt_run -x_tiles=2 -y_tiles=2  hello_world_many.c -ariane -finish_mask 0x1111111111111111 -rtl_timeout 10000000

Jbalkind, 2024-08-12:
Finish mask should be -finish_mask=1111 if you're running four cores - that could be the problem

ChiaLiu0618, 2024-08-12:
Hi, there seems to be the same issue.
[image]

Jbalkind, 2024-08-12:
Ok I think this is the same issue @guillemlp has been noting recently, which has been around for a long time. Multicore misbehaves with CVA6 and verilator but single core behaves correctly. Could we close this issue and open a new one for this specifically? I think we might have to dig into the CVA6 or OpenPiton code to try and figure out what's misbehaving.

closed 2024-08-13 by ChiaLiu0618