PrincetonUniversity/openpiton #151: How to port openpiton to a new FPGA?

LOCKEDGATE, 2024-07-22:
Hello,
How to port the openpiton to a new FPGA(xcvu19p), is there any official document? And how to make the corresponding test image?
Thanks.

Jbalkind, 2024-07-22:
To quote from a recent google group thread,

"There are some pointers in the fpga manual on openpiton.org but they're somewhat outdated. Worth checking out, though.

The main things are generating matching IP configurations for your board and figuring out top level pinout, reset polarity, clocking, switches. If you look in piton/tools/src/proto then you'll find that things are fairly cleanly separated by board for the Vivado scripting side. For the design side, you can look at the xilinx/ subfolders under e.g. piton/design/ and see that there'll be a folder per board to contain IP or the devices(_ariane).xml config files. You can also set relevant macros under your new ***_BOARD macro in piton/design/include/piton_system.vh

I'd encourage you to join the openpiton zulip (openpiton.zulipchat.com) where we can make a stream for this and give you some more direct assistance."

I believe someone has ported to xcvu19p FPGA parts before but the problem is that there tend to be different, bespoke boards which each need to be treated differently. They are also quite expensive so we don't have one in my team. Would still be happy to help with bringup, though.

LOCKEDGATE, 2024-07-23:
OK，I am in zulipchat，What should I do next?
[image]

Jbalkind, 2024-07-23:
I messaged you there - let's continue there

closed 2024-07-23 by Jbalkind