PrincetonUniversity/openpiton #152: Running a simulation error

ljh-num1, 2024-07-24:
When i run “sims -sys=manycore -x_tiles=1 -y_tiles=1 -vcs_run princeton-test-test.s” ， the error is as below：

‘’‘’‘’‘’‘’
[ICer@IC_EDA /home/ICer/openpiton/openpiton/build]$sims -sys=manycore -x_tiles=1 -y_tiles=1 -vcs_run princeton-test-test.s
sims -sys=manycore -x_tiles=1 -y_tiles=1 -vcs_run princeton-test-test.s 
sims: ====================================================
sims:   Simulation Script for OpenPiton
sims:   Modified by Princeton University on June 9th, 2015
sims: ====================================================
sims: ====================================================
sims:   Simulation Script for OpenSPARC T1
sims:   Copyright (c) 2001-2006 Sun Microsystems, Inc.
sims:   All rights reserved.
sims: ====================================================
sims: start_time Wed Jul 24 16:31:33 CST 2024
sims: running on IC_EDA
sims: uname is Linux IC_EDA 3.10.0-1160.83.1.el7.x86_64 #1 SMP Wed Jan 25 16:41:43 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
sims: version 2.0
sims: dv_root /home/ICer/openpiton/openpiton/piton
sims: model_dir /home/ICer/openpiton/openpiton/build
sims: tre_search /home/ICer/openpiton/openpiton/piton/tools/env/tools.iver
sims: using config file /home/ICer/openpiton/openpiton/piton/tools/src/sims/sims.config (-DFLIST_OST1)
sims: using random seed 0
sims: network_config not specified, assuming 2dmesh configuration
sims: Setting UART_DIV_LATCH to 0xb
sims: locating diag princeton-test-test.s
sims: Looking for diag under $DV_ROOT/verif/diag
sims: Found diag under /home/ICer/openpiton/openpiton/piton/verif/diag/assembly/princeton/princeton-test-test.s
sims: assembling diag
sims: midas -mmu=niagara -DPITON_X_TILES=1 -DPITON_Y_TILES=1 -DPITON_NUM_TILES=1 -DUART_DIV_LATCH=0xb -diag_root=/home/ICer/openpiton/openpiton/piton diag.s
Can't locate Bit/Vector.pm in @INC (@INC contains: /home/ICer/openpiton/openpiton/piton/tools/perlmod/BitFieldTie/1.09/lib/site_perl/5.8.0 /home/ICer/openpiton/openpiton/piton/tools/perlmod/Linux-x86_64 /home/ICer/openpiton/openpiton/piton/tools/perlmod //sun4-solaris / /home/ICer/openpiton/openpiton/piton/tools/perlmod/Linux-x86_64 /home/ICer/openpiton/openpiton/piton/tools/perlmod /home/ICer/openpiton/openpiton/piton/tools/perlmod/Midas/3.30/lib/site_perl/5.8.0 /usr/local/lib64/perl5 /usr/local/share/perl5 /usr/lib64/perl5/vendor_perl /usr/share/perl5/vendor_perl /usr/lib64/perl5 /usr/share/perl5 .) at /home/ICer/openpiton/openpiton/piton/tools/perlmod/BitFieldTie/1.09/lib/site_perl/5.8.0/BitFieldTie.pm line 30.
BEGIN failed--compilation aborted at /home/ICer/openpiton/openpiton/piton/tools/perlmod/BitFieldTie/1.09/lib/site_perl/5.8.0/BitFieldTie.pm line 30.
Compilation failed in require at /home/ICer/openpiton/openpiton/piton/tools/perlmod/TRELoad.pm line 98.
BEGIN failed--compilation aborted at /home/ICer/openpiton/openpiton/piton/tools/perlmod/Midas/3.30/lib/site_perl/5.8.0/Midas/Command.pm line 36.
Compilation failed in require at /home/ICer/openpiton/openpiton/piton/tools/perlmod/Midas/3.30/lib/site_perl/5.8.0/Midas/Interface.pm line 42.
BEGIN failed--compilation aborted at /home/ICer/openpiton/openpiton/piton/tools/perlmod/Midas/3.30/lib/site_perl/5.8.0/Midas/Interface.pm line 42.
Compilation failed in require at /home/ICer/openpiton/openpiton/piton/tools/perlmod/Midas/3.30/lib/site_perl/5.8.0/Midas.pm line 28.
BEGIN failed--compilation aborted at /home/ICer/openpiton/openpiton/piton/tools/perlmod/Midas/3.30/lib/site_perl/5.8.0/Midas.pm line 28.
Compilation failed in require at /home/ICer/openpiton/openpiton/piton/tools/perlmod/Midas/3.30/bin/midas line 29.
BEGIN failed--compilation aborted at /home/ICer/openpiton/openpiton/piton/tools/perlmod/Midas/3.30/bin/midas line 29.
sims: Caught a SIGDIE. midas compilation error at /home/ICer/openpiton/openpiton/piton/tools/src/sims/sims,2.0 line 2347.
‘’‘’‘’‘’‘’‘’‘’‘’‘’‘’‘’‘’‘’‘’‘’‘’‘’‘’‘’‘’‘’
“Can't locate Bit/Vector.pm in @INC (@INC contains“.
And if i use -msm_build and -msm_run. it  passed.

Jbalkind, 2024-07-24:
The bit vector package error is described in the readme, please install the associated package as it directs.

Also if it's passing with modelsim then you likely ran different commands, because this error is before the simulator runs.

ljh-num1, 2024-07-24:
After install perl-Bit-Vector.x86_64, the test is pass .
thanks.

closed 2024-07-24 by ljh-num1