PrincetonUniversity/openpiton #153: bram_test hello_world

ljh-num1, 2024-07-24:
Hi, 
I want to ensure that:
If i run the below three commands, the generated bit will print hello world on the uart console after reset? 
1  $MODEL_DIR:  sims -sys=manycore -x_tiles=1 -y_tiles=1 -msm_build -ariane
2  $MODEL_DIR:  sims -sys=manycore -msm_run -x_tiles=1 -y_tiles=1 hello_world.c -ariane -rtl_timeout 10000000
3  $PITON_ROOT: protosyn -b vcu118 -d system --core=ariane --bram-test hello_world
And the coe will put in which ram?  
Thanks!

Jbalkind, 2024-07-24:
Theoretically, yes. I will say that the bram-test flow hasn't been thoroughly tested for a while. The last time I tried it (I think on nexys4ddr), it printed but it also then ran some other code after the test finished. If you hit issues let me know and I'm happy to help debug!

ljh-num1, 2024-07-25:
[image]

What's this mean? Test failed!   
When protosyn , it will test hello_world?   
I already run sims , it passed.

ljh-num1, 2024-07-25:
The  logs are below:
[image]
The compilation.log is passed, the detailed command is as below:
[image]
The simulation.log is as below:
[image]
The results inicate that hello world is an unprocessed argument. So how can i use the argument --bram-test. Can you give me an example? 
thanks!

Jbalkind, 2024-07-25:
The test is probably called hello_world.c, not just hello_world?

ljh-num1, 2024-07-26:
I passed with the command "protosyn -b vcu118 -d system --core=ariane --bram-test hello_world.c".  Then i generated the project and bit .
I saw the prj didn't have mc and sd. Is that noraml?
Then i tested the bit. After loading bit, i changed sys_rst_n form 1->0. Then the uart print this:
[image]

When protosyn , this files are generated.  
[image]

And the fake_uart.log is as below, And I think after loading bit, it should print like :
[image]

ljh-num1, 2024-07-26:
Why it initialized sd? I think it should work from ram. Some switch should be selected?

ljh-num1, 2024-07-26:
I thind maybe the bootrom is error? it shouldn't initialize sd .  How can i change it or where can i get a right bootrom.v?

ljh-num1, 2024-07-26:
[image]
I find this ,but i don't know how to change it. I did't find some descriptions about this.

Jbalkind, 2024-07-29:
As I said, this configuration hasn't been tested for a long time. It was designed before the Ariane core was added and once we had DRAM and pitonstream working, we didn't have a lot of reason to use it. When I booted on nexys4ddr a design with --bram-test, it behaved weirdly, like it printed the test output and then ran the bootrom or something else that was strange like that.

I think probably you should check whether or not the PITONSYS_UART_BOOT or a similar macro is set. Either that or you should check the value of the uart_boot_en wire and change it for this configuration. That's probably what is controlling whether to use the linux bootrom or the bare metal bootrom.

ljh-num1, 2024-07-30:
ok, thanks.

closed 2024-07-30 by ljh-num1