PrincetonUniversity/openpiton #156: How to config many-nodes to work for openpiton?

zhb9103, 2024-09-19:
Hi, 
How can I config many-nodes to work for openpiton? And supported by current linux which is ariane-sdk(openpiton branch)?

Thanks!

Jbalkind, 2024-09-19:
Sorry could you be a little clearer about what you mean by many-nodes? Do you mean multiple chips connected together?

zhb9103, 2024-09-20:
Yes, I mean multiple chip connected together, as below:
[image]

The "node" concept comes from linux:
[image]

How can I do to impliment chips connection? 

Thanks!

Jbalkind, 2024-09-20:
Ok two things:
1. The original design relied on Coherence Domain Restriction (Fu et al. MICRO 2015) to enable multi-chip designs. That included ISA-specific components which have not been implemented for RISC-V yet.
2. Instead of using CDR, there is an enhanced design for this use case, SMAPPIC, built by Grigory Chirkov (Chirkov et al. ASPLOS 2023) which can handle multiple chip(let)s including on Amazon F1 FPGAs and is located here: https://github.com/PrincetonUniversity/smappic

You probably want to take a look at SMAPPIC. Note that it is not based on the openpiton git history so it will likely require a little tinkering if you want to take newer patches from OpenPiton and apply them there. SMAPPIC has a lot of great capabilities so I'd definitely recommend you check it out for your purposes.

zhb9103, 2024-09-21:
Got it. Thanks so much!

closed 2024-09-21 by zhb9103