PrincetonUniversity/openpiton #159: Facing Issue in accesing threads.

Rahulritgithub, 2025-01-09:
I'm Trying to access 4 threads each should calculate some factorial code parallely and should write in a specific directory
Using fopen() and fclose() file library

Jbalkind, 2025-01-09:
The bare metal environment used in simulation doesn't support functions like fopen() and fclose(). Are you able to run on FPGA instead?

Jbalkind, 2025-01-20:
I don't know. You haven't really given me enough information to say. This sounds a lot like a recent discussion on the openpiton Google group that you could read.

praveen1709-hub, 2025-01-22:
> I don't know. You haven't really given me enough information to say. This sounds a lot like a recent discussion on the openpiton Google group that you could read.

we are running on linux only so can we use fopen fclose?

Jbalkind, 2025-01-23:
Again, the core is running bare metal code. It does not have those functions.

praveen1709-hub, 2025-01-23:
> Again, the core is running bare metal code. It does not have those functions.

thanks for that, but is there any possible other ways you can suggest?

Jbalkind, 2025-01-23:
Read the thread I mentioned on the Google group.