PrincetonUniversity/openpiton #160: Need to print Real world time in a dummy log file

praveen1709-hub, 2025-01-22:
Is there any way to write the real world time in a dummy log file (fake_uart.log) with some string because i cant use any time.h or time() localtime()

Jbalkind, 2025-01-23:
The core doesn't know the real world time under simulation. A verilog system task like $time can give you the simulated time within the simulation I think. I have never looked but I'd guess there could be a verilog or systemverilog operator to give you the real time when the write in fake_uart.v is being performed. Worst case maybe you could use DPI to call into C++. This isn't a case I've ever thought about.

praveen1709-hub, 2025-01-23:
> The core doesn't know the real world time under simulation. A verilog system task like $time can give you the simulated time within the simulation I think. I have never looked but I'd guess there [...]

yeah, that the case i am assigned with so is there any other way you can possibly know pls share with me. we tried using clock cycle count to get_cycle_count fun to find the start time and stop time based on clock tick freq but it won't be a real time.

Jbalkind, 2025-01-23:
You could use SystemVerilog C++ DPI and bring in the time from C++ code. You'd need to figure out how to do this yourself.

praveen1709-hub, 2025-01-23:
ok thank you i will this 

> You could use SystemVerilog C++ DPI and bring in the time from C++ code. You'd need to figure out how to do this yourself.

praveen1709-hub, 2025-01-27:
> You could use SystemVerilog C++ DPI and bring in the time from C++ code. You'd need to figure out how to do this yourself.

we are trying to use systime in system time register mtime and mtimecmp is it possible in openpiton ?

Jbalkind, 2025-01-27:
Yes, there is a CLINT which contains mtime and mtimecmp registers which can be accessed via the memory app using load and store instructions (or the relevant CSR operations).

Also, this is a completely different question than what you asked before. The sense I get is that there are multiple of you working on the same team. I would appreciate if you could put your supervisor in touch with me by email so I can have a communication with them directly about the types of support requests I am receiving from your team.

praveen1709-hub, 2025-01-28:
> Yes, there is a CLINT which contains mtime and mtimecmp registers which can be accessed via the memory app using load and store instructions (or the relevant CSR operations).
>
> Also, this is a [...]

praveen1709-hub, 2025-01-28:
> Yes, there is a CLINT which contains mtime and mtimecmp registers which can be accessed via the memory app using load and store instructions (or the relevant CSR operations).
>
> Also, this is a [...]

sure i will ask them and tell you