PrincetonUniversity/openpiton #161: buidling simulation model using -vlt_build

ashuthosh-mr, 2025-03-05:
Hi,

I am trying to build ariane simulation model using the command: sims -sys=manycore -x_tiles=1 -y_tiles=1 -vlt_build -ariane

I get the error: bin/pyhp.py:140: SyntaxWarning: invalid escape sequence '\s'
  ws = re.compile("^(\s*)(.*)")
(many more lines like this and it fails) 

I am running it on Ubuntu 24.04
RISCV toolchain: 8.3 version SiFive
Verilator version: 5.035
I had manually installed spike and others that were mentioned in ariane_build_tools.sh

My suspects: I was not able to install python-pexpect which is a dependency. Instead I have python3-pexpect. I suspect this might be the reason why it fails. Please let me know if I am doing something wrong. Is installing python2 the only way ahead?

ashuthosh-mr, 2025-03-05:
[sims.log](https://github.com/user-attachments/files/19094152/sims.log) Including the sims.log for more details.

Jbalkind, 2025-03-06:
This sounds like a local python packaging issue rather than an openpiton issue specifically. Perhaps it's related to your python version? I believe pyhp.py was updated for python 3 but it's possible that the regex syntax used has changed with another python version?

ashuthosh-mr, 2025-03-07:
So I tried with different versions, 4.100, 4.200, 5.002. Different version gave different errors. I could share the sims.log for all the cases. But it would be helpful if you could point to the right version that might not give error. Thank you.

ashuthosh-mr, 2025-03-08:
With Verilator version 5.002, only error I had was timing which was solved by --no-timing. Late I came across another error with respect to init_jbus_model_call not being defined or found. I followed your verilator-5 branch with this commit: https://github.com/Jbalkind/openpiton/commit/7bbeec5a29e4139247bff6f7065303e98ede8105 

Added extern line. It worked. I got the output. :) 

Thanks a ton.

closed 2025-03-08 by ashuthosh-mr