PrincetonUniversity/openpiton #162: My CVA6-SDK Linux won't boot up

dassheladiya, 2025-03-19:
Hello,
I am writing to address an issue related to the CVA6-SDK. This matter is also documented as an open issue on the CVA6-SDK repository (issue: https://github.com/openhwgroup/cva6-sdk/issues/82). It has been suggested that this issue be raised on the OpenPiton repository.

I am currently running a dual-core CVA6 configuration on the Genesys2 board.

To boot Linux, I follow the steps came from segment: Booting SMP Linux on Genesys2 or VC707 in link https://github.com/PrincetonUniversity/openpiton
downloaded bbl.bin from: https://github.com/openhwgroup/cva6-sdk/releases/tag/v0.3.0-op

and burned it to the SD card as follows:
sudo sgdisk --clear --new=1:2048:67583 --new=2 --typecode=1:3000 --typecode=2:8300 /dev/sdb
sudo dd if=bbl.bin of=/dev/sdb1 oflag=sync bs=1M

Then I inserted the SD card into the FPGA development board and it ran fine.

After that I wanted to compile a version under CVA6-SDK for testing.

I did all the steps under ubuntu as described in Quickstart:

step 1. sudo apt-get install autoconf automake autotools-dev curl libmpc-dev libmpfr-dev libgmp-dev libusb-1.0-0-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev device-tree-compiler pkg-config libexpat-dev

step 2. git submodule update --init --recursive

step 3. make images

At the end of the compilation, I can see the corresponding files generated in the /home/cva6-sdk/opensbi/build/platform/fpga/openpiton/firmware:

fw_payload.bin fw_payload.elf 

Then I have burned it to the SD card:

step 4. sudo dd if=/home/cva6-sdk/opensbi/build/platform/fpga/openpiton/firmware/fw_payload.elf of=/dev/sdb1 status=progress oflag=sync bs=1M

After that, I inserted the SD card into the FPGA development board and restart board but after loading block, system is blocked with the following message:

----------------------------------------
--     OpenPiton+Ariane Platform      --
----------------------------------------
OpenPiton Version: b'd0093384'                   
Ariane Version:    b'4c01614f'                   
                                        
FPGA Board:        genesys2                   
Build Date:        Mar 10 2025 15:20:14                   
                                        
#X-Tiles:          2                   
#Y-Tiles:          1                   
#Cores:            2                   
Core Freq:         66 MHz                   
Network:           2dmesh_config                   
DRAM Size:         1024 MB                
                                        
L1I Size / Assoc:   16 kB / 4          
L1D Size / Assoc:    8 kB / 4          
L15 Size / Assoc:    8 kB / 4          
L2  Size / Assoc:   64 kB / 4          
----------------------------------------

sd initialized!
initializing SD... 
sd initialized!
copying block 0 of 1 blocks (0 %)
gpt partition table header:
	signature:	305A10732E4A0A13
	revision:	00035A17
	size:		EE0A0A13
	crc_header:	01336A89
	reserved:	0433015A
	current lba:	0933000584B30005
	backup lda:	0A33000689B30006
	partition entries lba:   	89B3000609330005
	number partition entries:	0A330006
	size partition entries:  	00EF0007
copying block 0 of 1 blocks (0 %)
gpt partition entry 00
	partition type guid:	00000000000000000000000000000000
	partition guid:     	00000000000000000000000000000000
	first lba:	0000000000000000
	last lba:	0000000000000000
	attributes:	0000000000000000
	name:	000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
gpt partition entry 01
	partition type guid:	00000000000000000000000000000000
	partition guid:     	00000000000000000000000000000000
	first lba:	0000000000000000
	last lba:	0000000000000000
	attributes:	0000000000000000
	name:	000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
gpt partition entry 02
	partition type guid:	00000000000000000000000000000000
	partition guid:     	00000000000000000000000000000000
	first lba:	0000000000000000
	last lba:	0000000000000000
	attributes:	0000000000000000
	name:	000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
gpt partition entry 03
	partition type guid:	00000000000000000000000000000000
	partition guid:     	00000000000000000000000000000000
	first lba:	0000000000000000
	last lba:	0000000000000000
	attributes:	0000000000000000
	name:	000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
copying boot image 
copying block 65500 of 65536 blocks (99 %)
 done!

Jbalkind, 2025-04-03:
The regular cva6-sdk isn't designed to run with OpenPiton. You could try this branch instead which I have kept more up to date and tends to work better: https://github.com/Jbalkind/ariane-sdk/tree/openpiton-opensbi

That said, it's also important to make sure you're using a new enough FPGA bitstream (the five year old ones that are posted won't work with newer Linux+opensbi). Sometimes the SD controller can also be picky about SD cards and it might take trying a couple of different ones before seeing success. Not exactly sure why.

dassheladiya, 2025-04-04:
Hello Jonathan,
Thank you very much for pointing out the correct branch. I tried using several SD cards, but they did not work. Consequently, I generated a new bit stream in which I modified the bootrom code to remove the SD copy part. Additionally, I mapped the JTAG signals to PMOD to connect the Lauterbach debugger. I then loaded fw_payload.bin at 0x080000000 using the Lauterbach debugger. Now, I am able to boot Linux, which I generated using your branch (https://github.com/Jbalkind/ariane-sdk/tree/openpiton-opensbi).